nccgroup/Sniffle #2: Installation Process

anini9519, 2019-11-22:
Is there a video that I can refer to for the Installation and Configure Sniffle ? 
ps:Apart from the steps written in the readme file.

sultanqasim, 2019-11-22:
I don't have a video of the process, though perhaps I could make one. Where are you having issues?

If you're having difficulty building the firmware, you could use the prebuilt firmware binaries here (just flash them with TI UniFlash): https://github.com/nccgroup/Sniffle/releases/tag/v1.1

For the host side Python code, you don't need to install anything beyond Python 3.5+ and PySerial.

anini9519, 2019-11-22:
Thank you for your response.
I was trying to install the firmware and I get this output from the terminal.
It would really great if you can make a video 
[image]

sultanqasim, 2019-11-22:
Looking at your image, does the path it's complaining about exist? You might need to edit this line: https://github.com/nccgroup/Sniffle/blob/master/fw/makefile#L1

If building the firmware is too much work in your setup, you can just use the prebuilt firmware I linked earlier.

anini9519, 2019-11-25:
I was able to flash the prebuilt firmware to TI board.
Still, It would be really great if you give me steps in more elaborate way to install the firmware from scratch.
Can you give me steps how to sniff ble packets with the given command line ?
Thanks in Advance.

sultanqasim, 2019-11-25:
Read the README.

closed 2019-11-25 by sultanqasim