nccgroup/Sniffle #3: Firmware freezing issue

jakwnd, 2019-12-18:
I am using a LAUNCHXL-CC26X2R1 with a CC2652R1, I cannot get the firmware to stay stable for any filtering. The board freezes if I use any filter. If I reset the board it will capture packets but does not apear to reflect the filters, All I can see are advertisements.

Is there anything I can do to make it more stable?

sultanqasim, 2020-02-10:
I haven't observed any issues with freezing caused by filtering myself, but I have had a few cases of the board freezing by itself regardless of filter configuration, usually when the device boots through power-on reset (POR). I couldn't find any issues in the code for this, but I suspect it might be insufficiently stable supply voltage during initialization causing some odditites.

Try this:
1. Plug the sniffer into your computer.
2. Wait a few seconds, then press the reset button (supply voltage should be stable now).
3. Start the host-side software to configure and operate the sniffer after you reset the board.

sultanqasim, 2020-02-16:
I have been doing some more experimenting with the freezing bug, but I've been having a terrible time reproducing the issue - I have reproduced a freeze maybe 3 or 4 times, but that very was over 100+ plug/unplug attempts and a few hours.

sultanqasim, 2020-02-16:
Another thing you could do if you can't press the reset button before you first launch the sniffer would be to use this script with the lastest firmware: https://github.com/nccgroup/Sniffle/blob/master/python_cli/reset.py

It's not foolproof, but it usually revives the firmware if it's half alive.

jakwnd, 2020-02-18:
Im sorry for ghosting a bit here, I dont have the board w/ me atm but I had much better luck at home. My office environment seemed to be too busy. 

I will bring it in tomorrow and reproduce the issues and try to get you more info.

breakersun, 2020-11-26:
I am using a LAUNCHXL-CC26X2R1 too.
When I tried to capture long-range extended advertisement with "scanner.py -s COM15 -e -l", the board freezing and the red led is off.
My office is a bit of noisy in 2.4Ghz radio space.

sultanqasim, 2021-07-13:
@breakersun The red LED only blinks when the board is receiving something; it won't blink when you're trying to listen to long range advertisements if it doesn't see any. Does performing a regular (non-long-range) scan afterwards work normally? If regular (1M PHY) scanning afterwards works, the board isn't frozen, and if 1M PHY scanning sees nothing even with advertising devices nearby, it's actually frozen.

breakersun, 2021-07-16:
@sultanqasim thank you for the info.
regular scan works fine, so it's not actually frozen. it just never returns after starting a round of "long-range extended" scan.

sultanqasim, 2021-07-22:
OK. At least on firmware version 1.6, it works fine for me. I can scan long range, see the results if anything is advertising on the long range PHY, and then switch to the basic 1M PHY for normal scanning and that works too. Try out the latest version, and let me know if it works for you.

sultanqasim, 2022-04-05:
Closing as stale because I haven't encountered any firmware freezes in the last year, and I haven't heard any new reports or updates from prior reporters.

closed 2022-04-05 by sultanqasim