nccgroup/Sniffle #4: using alone module

jurek4321, 2020-02-15:
Is it possible to use alone CC2652R module without having launch board?
I found this on ebay www.ebay.com/itm/1-8-3-3V-CC2652R-Bluetooth-5-0-ZigBee-Module-2-4GHZ-Wireless-Module-Low-Power/333451099943

If its possible how to program it?

sultanqasim, 2020-02-16:
It's definitely possible to use it on module rather than a Launchpad board. Compile with PLATFORM=CC2652R1F just as you would on the Launchpad. You just need to specify the desired PIN assignment for UART in the SysConfig file:

- https://github.com/nccgroup/Sniffle/blob/master/fw/sniffle.syscfg#L54
- https://github.com/nccgroup/Sniffle/blob/master/fw/sniffle.syscfg#L67

jurek4321, 2020-02-21:
Which programmer i need to program it? I see some cheap "cc debuggers"  (SmartRF04EB) on aliexpress are they good to program it?

sultanqasim, 2020-02-24:
XDS110 (either standalone or built into a Launchpad) is what TI recommends, but in theory any ARM microcontroller debugger should work with the right configuration. There are also cheaper boards with XDS110 on them, such as CC-DEVPACK-DEBUG, but you'll have to make the connections work.

jurek4321, 2020-03-09:
I was searching for cheap programmers and the cheapest i found is this: https://pl.aliexpress.com/item/32963210375.html but this module with programmer gives me price around price of launchpad board. Is there any cheaper programmers or any other way to program it using maybe arduino or LPT port?

jurek4321, 2020-03-09:
Are the "ARM microcontroller debugger" should have any specific requirements or every should work?
Is there any TI tested or supported debuggers list?

sultanqasim, 2020-03-11:
XDS110 is made by TI and known to work. Other debuggers may be able to work with appropriate configuration, but I haven't tried.

What are you trying to accomplish? Do you just want the cheapest path to a single working Sniffle device, or are you trying to mass produce Sniffle-compatible devices? If you just want a single working Sniffle, you may as well buy the Launchpad if you put any value on your time. If you want to mass produce devices, it's a different matter, and you can use a standalone off-board programmer to avoid the BOM cost of including the programmer.

jurek4321, 2020-03-12:
I just dont want to pay $50 for sniffle, XDS110 is good but if i will using it only for sniffle i will probably use it only one time so i dont want to use programmer that cost more than module.

sultanqasim, 2020-03-12:
By the time you solder a UART to USB adapter to the module, and get a cheaper programmer to work, the time you'd have spent would be worth more than the minimal extra cost of just buying a Launchpad.

closed 2020-03-12 by sultanqasim