nccgroup/Sniffle #5: No output from Sniffer

manuHimself, 2020-02-18:
I just setup the project as described on a CC26X2R1 by following the instructions but I'm not able to get any output from the board. I tried both, Win10 and Ubuntu and building & flashing seem to work.
Any suggestion, what could went wrong here?

Thanks!

sultanqasim, 2020-02-18:
When you press thre reset button on the board beside the USB port, do you get any blinking of the red LED? If not, the firmware was probably built or flashed incorrectly. I'm going to be releasing new prebuilt firmware binaries tagged as version 1.2 later today, so you can try flashing those.

sultanqasim, 2020-02-18:
Version 1.2 has now been tagged, and prebuilt binaries are available. See the releases page: https://github.com/nccgroup/Sniffle/releases

manuHimself, 2020-02-19:
Thanks! I get some blinking of the red LED.
I flashed your last pre-build binary but still gettting the same behaviour. 
Maybe it's more related to the serial output. I'll double check my pyserial installation...

sultanqasim, 2020-02-24:
Every time the red LED blinks, that means a BLE frame has been received over the air and transmitted to the host over UART. It could be a PySerial issue, Linux kernel issue, or a jumper issue. Make sure the RX/TX jumpers between the debugger and CC2652 are in place.

manuHimself, 2020-02-26:
I retried and within Windows it's working now. Thanks for the help! Great tool!

closed 2020-02-26 by manuHimself