nccgroup/Sniffle #8: Support for CC2640R2?

Danyc0, 2020-02-21:
Do you think Sniffle could theoretically support the CC2640R2?

Thanks!

sultanqasim, 2020-02-24:
Yes, it's possible to modify it to support the older CC2640R2, I just haven't tried and don't currently plan on doing so.

Danyc0, 2020-02-24:
That's great, thank you! I only ask because the CC2640R2 dev board is half the price of the currently supported ones.

sultanqasim, 2020-02-24:
I had another user ask for this too, and a friend/colleague in the company who has a few CC2640R2F boards also wants this. If you want to add support for it, be my guest. You'll need to:

- Create a platform-specific subdirectory for the CC2640R2 in the firmware directory
- Generate a CCXML file for the CC2640R2F using Uniflash
- Copy over the linker file (.lds) and *_LAUNCHXL_fxns.c file from SDK examples for the CC2640R2F
- Edit the XDC CFG file to use the CC2640R2 ROM for that platform: https://github.com/nccgroup/Sniffle/blob/master/fw/sniffle.cfg#L345
- Edit the makefile to use the CC2640R2 SDK instead of the CC13x2/CC26x2 when the PLATFORM variable is set to the CC2640R2F
- Add other platform specfic options to the makefile as necessary, and move CCxx2 common things in the makefile back to the platform specific "if" sections
- If I'm using any CCxxx2 specific peripherals or hardware features anywhere, work around those. There shouldn't be much if anything that's not supported on the CC2640R2, as the radio functionality is mostly the same, but there might be some minor differences not coming to mind immediately.

PRs welcome if you have the time :)

Danyc0, 2020-02-26:
I had some time so I had a look at porting it, but sysconfig is running into issues with the radioconfig stuff, any ideas? Feel free to check out my changes: [https://github.com/Danyc0/Sniffle](url)

Build log:
> Generating configuration files...
Running script...
Error: Exception occurred calling scripting.addModule(): No such resource: /ti/devices/radioconfig/custom.syscfg.js
Paths searched: 
/home/dan/ti/simplelink_cc2640r2_sdk_3_40_00_10/source/ti/devices/radioconfig/custom.syscfg.js
/home/dan/ti/simplelink_cc2640r2_sdk_3_40_00_10/source/ti/devices/radioconfig/.meta/custom.syscfg.js
/home/dan/ti/simplelink_cc2640r2_sdk_3_40_00_10/source/ti/devices/.meta/radioconfig/custom.syscfg.js
/home/dan/ti/simplelink_cc2640r2_sdk_3_40_00_10/source/ti/.meta/devices/radioconfig/custom.syscfg.js
/home/dan/ti/simplelink_cc2640r2_sdk_3_40_00_10/source/.meta/ti/devices/radioconfig/custom.syscfg.js
/home/dan/git/sniffle-dan/fw/ti/devices/radioconfig/custom.syscfg.js
/home/dan/git/sniffle-dan/fw/ti/devices/radioconfig/.meta/custom.syscfg.js
/home/dan/git/sniffle-dan/fw/ti/devices/.meta/radioconfig/custom.syscfg.js
/home/dan/git/sniffle-dan/fw/ti/.meta/devices/radioconfig/custom.syscfg.js
/home/dan/git/sniffle-dan/fw/.meta/ti/devices/radioconfig/custom.syscfg.js
    at Object.r [as addModule] (/home/dan/ti/sysconfig_1.3.1030/dist/cli.js:33:27617)
    at eval (/home/dan/git/sniffle-dan/fw/sniffle.syscfg:42:22)
    at n.each (/home/dan/ti/sysconfig_1.3.1030/dist/cli.js:26:578809)
    at Wt (/home/dan/ti/sysconfig_1.3.1030/dist/cli.js:9:5239)
    at Function.Ga (/home/dan/ti/sysconfig_1.3.1030/dist/cli.js:9:40286)
    at Object.t.runScript (/home/dan/ti/sysconfig_1.3.1030/dist/cli.js:26:578693)
    at <anonymous>
make: *** [makefile:167: syscfg] Error 1

sultanqasim, 2020-02-26:
Probably the way to declare BLE PHY for sysconfig is different on the CC2640. Try using the SysConfig GUI tool to generate a syscfg file for CC2640R2F with the BLE5 PHY, and then seeing how it declares it. You can then modify the syscfg file by adding an if clause around the lines that need to be different. You'll need to do some reading figure out what the if clause should actually test (probably system.deviceInfo._something_).

Danyc0, 2020-02-28:
So it looks like the CC2640R2F isn't actually supported by SysConfig, I've reached out to TI and there's no word from them yet about when/if it will ever be supported. Even without the official support it seems to have some basic functionality, but nothing like what can be done with the CC26x2 chips, for example there's no section of configuring it for BLE or BLE5(custom).

sultanqasim, 2020-02-28:
In that case, as a crude workaround, you could run SysConfig for the CC2652R, and manually edit the generated ti_radio_config.c/ti_radio_config.h to suit the CC2640R2. For the other SysConfig generated files (not radio related), you could use what SysConfig already generates for the CC2640R2.

After that, just edit the makefile to not run sysconfig, and remove them from .gitignore, so that your manually created files persist. Not a nice way for merging upstream, but if you can at least get it working, that would be a great start.

sultanqasim, 2021-07-29:
I have no plans for this, and newer chip variants keep coming out, so closing.

closed 2021-07-29 by sultanqasim