nccgroup/Sniffle #11: MacsO Catalina "arm-none-eabi-gcc" cannot be opened because the developer cannot be verified

bobtaylorlsi, 2020-03-30:
Hi, tried this on my Mac and it failed - here is the screen dump of the attempt to make.

bobtaylor@Bobs-MacBook-Pro Sniffle-master % cd fw
bobtaylor@Bobs-MacBook-Pro fw % make
Running Configuro...
Unable to find any JVMs matching version "1.7".
making package.mak (because of package.bld) ...
gnu.targets.arm.M4F: warning: failed to parse '-dumpversion' output (= ''); check compiler options
gnu.targets.arm.M4F: warning: can't determine gcc version; 
Check if /Users/bobtaylor/ti/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc can run on 
MacOS. 
Also, try explicitly setting the GCCVERS configuration parameter.
gnu.targets.arm.M4F: warning: failed to parse '-v' output (= ''); check compiler options
gnu.targets.arm.M4F: warning: can't determine binutils version; 
Check if /Users/bobtaylor/ti/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-ld can run on 
MacOS. 
Also, try explicitly setting the BINVERS configuration parameter.
generating interfaces for package rtos_build (because package/package.xdc.inc is older than package.xdc) ...
configuring sniffle.xm4fg from package/cfg/sniffle_pm4fg.cfg ...
generating custom ROM library makefile ... 
Starting build of library sources ...
making /Users/bobtaylor/ti/Sniffle-master/fw/rtos_build/package/cfg/sniffle_pm4fg.src/sysbios/rom_sysbios.am4fg ...
asmm4fg /Users/bobtaylor/ti/simplelink_cc13x2_26x2_sdk_3_40_00_02/kernel/tirtos/packages/ti/sysbios/family/arm/m3/Hwi_asm_gnu.sv7M ...
gmake[1]: *** [makefile:49: m3_Hwi_asm_gnu.o] Killed: 9
gmake: *** [/Users/bobtaylor/ti/Sniffle-master/fw/rtos_build/package/cfg/sniffle_pm4fg.src/makefile.libs:57: /Users/bobtaylor/ti/Sniffle-master/fw/rtos_build/package/cfg/sniffle_pm4fg.src/sysbios/rom_sysbios.am4fg] Error 2
Build of libraries failed.
error: xdc.cfg.SourceDir: "/Users/bobtaylor/ti/xdctools_3_60_02_34_core/packages/xdc/cfg/SourceDir.xs", line 209: xdc.cfg.SourceDir : Build of generated source libraries failed: exit status = 2:

js: "/Users/bobtaylor/ti/xdctools_3_60_02_34_core/packages/xdc/cfg/Main.xs", line 160: Error: Configuration failed!
gmake: *** [package.mak:218: package/cfg/sniffle_pm4fg.xdl] Error 1
gmake: *** Deleting file 'package/cfg/sniffle_pm4fg.xdl'
gmake: *** [package/cfg/sniffle_pm4fg.xdl] Deleting file 'package/cfg/sniffle_pm4fg.h'
gmake: *** [package/cfg/sniffle_pm4fg.xdl] Deleting file 'package/cfg/sniffle_pm4fg.c'
js: "/Users/bobtaylor/ti/xdctools_3_60_02_34_core/packages/xdc/tools/Cmdr.xs", line 52: Error: xdc.tools.configuro: configuration failed due to earlier errors (status = 2); 'linker.cmd' deleted.
make: *** [rtos_build/linker.cmd] Error 1
bobtaylor@Bobs-MacBook-Pro fw %

sultanqasim, 2020-03-30:
I haven't yet upgraded any of my Macs to Catalina, but I assume this is some binary signing issue with the compiler impacting Gatekeeper. Not much I can do in Sniffle about this, but maybe there's some setting to disable signature verification on Catalina to make the compiler work. When I get around to upgrading my Macs to Catalina, I can look into this, but for now the easiest approach may be to just build it in a Linux VM.

sultanqasim, 2020-03-30:
Which specific board are you targeting? If you don't have time to set up a build environment, I can just make a build of the latest firmware for you.

bobtaylorlsi, 2020-03-31:
Hi, I am setting up on my Linux Ubuntu machine. The Mac is a pain - I use it for iOS development and run parallels for Windows to run Alitum for PCB layout. If you have time to build the firmware that is great. We are working during the COVID-19 remote and office (split) as an essential industry here in Ohio. I am in the middle of developing a Bluetooth MESH 5.x lighting control system - hence the need for Sniffle. I have the either the CC2652RB, CC1352R1, or CC26x2R1in my lab.

sultanqasim, 2020-04-02:
Attached is a zip with builds of the latest master branch (e5297e2be967db617e8f42a20107d34b5eddc56d):
[sniffle_master_builds.zip](https://github.com/nccgroup/Sniffle/files/4422952/sniffle_master_builds.zip)

I've never done anything with Bluetooth Mesh before, but my understanding is that it builds upon regular Bluetooth Low Energy, so hopefully this will do what you want.

sultanqasim, 2020-04-24:
Not a Sniffle issue, and prebuilt binaries are available on the releases tab for Version 1.3 released today.

closed 2020-04-24 by sultanqasim