nccgroup/Sniffle #12: How to use standalone SysConfig to generate files?

homewsn, 2020-05-02:
Standalone desktop variant of TI SysConfig (1.4.0+1234) tool for Windows doesn't generate source files from the sniffle.syscfg file. I get an error message: 

> An attempt to set this to the invalid value cmdBle5GenericRx,cmdBle5Master,cmdBle5RadioSetup,cmdBle5Slave,cmdBleAdv,cmdBle5Initiator,cmdBle5Scanner,cmdFs was blocked. This is most likely due to a backwards compatiblity discrepancy. You can dismiss this error by either accepting the current value, or changing it to another one.

 If I accept the current value files are generated but RF_cmdBle5Master etc. are undefined.
The screenshot is attached below:
[image]

How did you manage to get SysConfig to generate files? Maybe versions for Windows and for Linux are different? 
Could you please shared your SysConfig generated files?

sultanqasim, 2020-05-02:
I've never tried building on Windows, though it should be possible. I've only compiled it on Linux. On Linux, when following the steps in the README, running make should automatically invoke the syscfg CLI to generate the necessary files. See this line of makefile for how it's invoked: https://github.com/nccgroup/Sniffle/blob/master/fw/makefile#L138

Which platform are you building for? CC2642R/CC2652R, CC2652RB, or CC1352R? In the interim, I can generate the files for you on Linux targeting the correct platform, or you can just use the prebuilt firmware on the releases tab.

homewsn, 2020-05-03:
I am using Sniffle v. 1.1 on the custom CC1352P1 module (equals to LAUNCHXL-CC1352P1). The firmware has been built and works without problems. Could you please generate files for CC1352P1? If not, the files for CC1352R will be fine too.

homewsn, 2020-05-03:
I have tried standalone and online version of SysConfig. The both can open sniffle.syscfg and generate files with option
`// @cliArgs --product "simplelink_cc13x2_26x2_sdk@4.10.00.78" --board "ti/boards/CC1352R1_LAUNCHXL"`
but option
`// @cliArgs --product "simplelink_cc13x2_26x2_sdk@4.10.00.78" --board "ti/boards/CC1352P1_LAUNCHXL"`
leads to an error.
I am not sure but it seems that the files generated for CC1352R1_LAUNCHXL are suitable for CC1352P1_LAUNCHXL.

homewsn, 2020-05-03:
I want to confirm that the files created for CC1352R1_LAUNCHXL are suitable for CC1352P1_LAUNCHXL, at least in the sniffer part. I was able to successfully build and flash the firmware to my custom CC1352P board. Thank you.
But I still have no idea why I may not use the option `--board "ti/boards/CC1352P1_LAUNCHXL"`. Maybe you have an idea?

sultanqasim, 2020-05-04:
There might be an error in TI's sysconfig meta json files within the SDK. I'll examine this when I have more time.

homewsn, 2020-11-07:
[Replacing](https://github.com/nccgroup/Sniffle/issues/24#issuecomment-712591485) cmdBle5RadioSetup with cmdBle5RadioSetupPa in sniffle.syscfg works for me. Thank you.
I would like to point out that [using the files created by SysConfig for CC1352R1_LAUNCHXL in the build for CC1352P1 custom board](https://github.com/nccgroup/Sniffle/issues/12#issuecomment-623114431) ultimately leads to unstable sniffer operation and missing packets.

closed 2020-11-07 by homewsn