nccgroup/Sniffle #15: Missing Mac Address in Connection

TorstenRobitzki, 2020-05-08:
I traced the advertisements of a specific device by using the MAC address filter:

    ./sniff_receiver.py -s /dev/cu.usbmodemL1100GTV1 -m d9:76:fd:15:1a:e3 -o 1249_double.pcap

Viewing the pcap file in Wireshark results in having the PDU source and destination being filled with `Unknown_0x50655bda`, for both, master and slave, while the adverting data and the connection request where filled with the with correct packet source and destination.

sultanqasim, 2020-05-08:
This is a limitation of the current PCAP packet format for BLE RF. It currently provides no means to indicate the direction. I’m working on improving the packet format together with other users of it so that the PCAP can include information about direction, PHY, and some BLE 5 related aspects. https://code.wireshark.org/review/#/c/37142/

sultanqasim, 2020-07-10:
The pcap_new branch labels the PHY in use, and indicates direction for data packets using the revised DLT/LINKTYPE spec here: https://gistcdn.githack.com/sultanqasim/8b6561309f5934f084a0d938ae733b7a/raw/c9172a730117c824a1b80add472052220810e538/LINKTYPE_BLUETOOTH_LE_LL_WITH_PHDR.html

I'm working with the tcpdump.org folks to make this spec update official. After that, the relevant Wireshark change can be merged.

sultanqasim, 2020-07-13:
Wireshark change merged, Sniffle changes merged, just waiting for this PR to be merged before I mark this as closed: https://github.com/the-tcpdump-group/tcpdump-htdocs/pull/17

sultanqasim, 2020-07-13:
Spec update is now official

closed 2020-07-13 by sultanqasim