nccgroup/Sniffle #17: Is ADV_EXT_IND with an additional AUX_CHAIN_IND supported ?

piotrwiniarczyk-silvair, 2020-07-01:
Hi

I am seeing this:
Timestamp: 2.140122     Length: 15      RSSI: -64       Channel: 37     PHY: 1M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 13
AuxPtr Chan: 5 PHY: 2M Delay: 191250 us
AdvMode: Non-connectable, non-scannable
AdvA: 4A:C9:C9:2E:B2:B0 (RPA) AdvDataInfo: F2 0D
47 0D 0C 19 B0 B2 2E C9 C9 4A F2 0D 05 E7 38
G                 .        J              8

Timestamp: 2.331504     Length: 257     RSSI: -66       Channel: 5      PHY: 2M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 255
**AuxPtr Chan: 6 PHY: 2M Delay: 2490 us**
AdvMode: Non-connectable, non-scannable
AdvDataInfo: F2 0D
47 FF 06 18 F2 0D 06 53 20 5D 16 45 23 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 5D 16 56 34 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 5D 16 34 12 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66 77 88 99 00 11 22 33 44 55 66

But I am still expecting one AUX_CHAIN_IND packet after ADV_EXT_IND.
I use cc1352r1 as sniffer and S10e mobile phone as advertiser with nRF Connect app.

sultanqasim, 2020-07-01:
AUX_CHAIN_IND is not explicitly *not* supported, but I hadn’t tested it, and probably should. I’ve been meaning to make a semi-automated test suite for Sniffle at some point. My guess as to what’s happening is that it’s not spending enough time on the aux channel to get the second AUX_CHAIN_IND frame. If the initial AUX_ADV_IND was shorter, it might capture it.

I do intend to properly support AUX_CHAIN_IND, so I’d consider this a bug to be fixed. I may have some free time on my schedule next week, so I may look into this and the other bug reports then.

sultanqasim, 2020-07-01:
Try this: change `auxPeriod` from 4000 * 4 to 1500 * 4 (and recompile the firmware of course). It may fix your issue, though I haven't verified.

https://github.com/nccgroup/Sniffle/blob/master/fw/RadioTask.c#L979

piotrwiniarczyk-silvair, 2020-07-01:
I have try other PHYs. No success.
For coded+coded or 1M+coded the delay is longer, but still no AUX_CHAIN_IND.

Since your code is the only AE sniffer I have an access to I cannot confirm 100% that S10E mobile is things properly. 

I have also filled a Wireshark bug for parsing such ADV_EXT_IND packets. 
https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=16666

BTW. The Wireshark has quite few different Bluetooth packets formats and WTAP_ENCAP_NORDIC_BLE has probably all you needed (direction and phy-s). Expanding other capture format is also also a good idea.

---
Timestamp: 22.355955    Length: 9       RSSI: -73       Channel: 37     PHY: Coded
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 35 PHY: Coded Delay: 60000 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo: 42 0E
47 07 06 18 42 0E 23 D0 47
G           B     #     G

Timestamp: 22.415939    Length: 257     RSSI: -74       Channel: 35     PHY: Coded
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 255
AuxPtr Chan: 36 PHY: Coded Delay: 4980 us
AdvMode: Non-connectable, non-scannable
AdvA: 64:65:78:E8:01:E9 (RPA) AdvDataInfo: 42 0E

sultanqasim, 2020-07-01:
I'm pretty sure the S10e is transmitting AUX_CHAIN_IND correctly. The issue is the scheduling of channel hopping. It was spending too much time on the channel of the AUX_ADV_IND, and so not hopping to the channel of the next AUX_CHAIN_IND in time. The S10e is using a fairly short (but still permissible) delay that is too short for the currently set auxPeriod to tolerate. This is why I suggested reducing auxPeriod. It should still listen to packet completion when it goes beyond auxPeriod before moving onto the next channel.

sultanqasim, 2020-07-01:
Regarding packet decoding, the path I want to take is this: https://code.wireshark.org/review/#/c/37142/

I just need to get the spec change finalized and approved by others. I hadn't been pursuing it for a few weeks because I got busy with other work, but hopefully I'll have some time next week.

The Nordic BLE format supports the necessary decoding features, but the format is very specific to the firmware and behaviour of the Nordic sniffer, which is why I wanted to create a more specific format. There was also a long email thread going on the side of the gerrit where we (myself, Joakim, Guy, and Mike Ryan) discussed and improved my proposed format changes.

sultanqasim, 2020-07-01:
When I wrote the scheduling logic, I had mistakenly assumed AUX_CHAIN_IND would come on the same channel as the AUX_ADV_IND, hence the long auxPeriod I had used, but that was a bad assumption.

sultanqasim, 2020-07-01:
Also, if reducing auxDelay doesn't suffice to make AUX_CHAIN_IND capture work, it may be due to RTOS latency issues, and I may need to make some changes to this logic to reduce channel hopping latency: https://github.com/nccgroup/Sniffle/blob/master/fw/RadioTask.c#L986

piotrwiniarczyk-silvair, 2020-07-01:
No luck with 1500 * 4 or even 800 * 4. Below 800 * 4 the ADV_EXT_IND on data channel stooped to show.

sultanqasim, 2020-07-01:
Thanks for trying it out. It's probably an RTOS latency issue with channel hop timing (still getting onto the channel for the AUX_CHAIN_IND a little too late to grab the sync word). I have an idea on how to fix it, but it'll take a little more work than just changing a coefficient.

sultanqasim, 2020-07-09:
I put together an Android app that sends long extended advertisements requiring AUX_CHAIN_IND, and tested it on my Samsung Galaxy S10e. Sniffle was indeed broken in receiving AUX_CHAIN_IND, but I was able to fix it in the way I expected by just reducing auxPeriod the way I suggested in an earlier comment. Not sure why it didn't work for you, but it works for me with the change. See the example below. The Python host side code doesn't currrently distinguish ADV_EXT_IND vs AUX_ADV_IND vs AUX_CHAIN_IND, but that's a different issue.

```
Timestamp: 10.146471    Length: 15      RSSI: -69       Channel: 37     PHY: 1M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 13
AuxPtr Chan: 36 PHY: 2M Delay: 46230 us
AdvMode: Non-connectable, non-scannable
AdvA: 73:9C:85:A2:F4:EE (RPA) AdvDataInfo: 42 03
47 0D 0C 19 EE F4 A2 85 9C 73 42 03 24 05 26
G                          s  B     $     &

Timestamp: 10.192731    Length: 257     RSSI: -72       Channel: 36     PHY: 2M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 255
AuxPtr Chan: 11 PHY: 2M Delay: 2490 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo: 42 03
47 FF 06 18 42 03 0B 53 20 0C 09 47 61 6C 61 78 79 20 53 31 30 65 03 03 1D B8 B3 16 1D B8 01 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
G           B        S           G  a  l  a  x  y     S  1  0  e                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 

Timestamp: 10.195231    Length: 257     RSSI: -66       Channel: 11     PHY: 2M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 255
AuxPtr Chan: 12 PHY: 2M Delay: 2490 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo: 42 03
47 FF 06 18 42 03 0C 53 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
G           B        S                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           

Timestamp: 10.197731    Length: 257     RSSI: -65       Channel: 12     PHY: 2M
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 255
AuxPtr Chan: 13 PHY: 2M Delay: 2490 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo: 42 03
47 FF 06 18 42 03 0D 53 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
G           B                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       ```

sultanqasim, 2020-07-09:
After some more testing, one limitation I discovered is that if the gap between end of the last aux packet and the start of the next one (on a different channel) is less than roughly 400 us, Sniffle will fail to receive the next aux packet because the latency in processing the received packet and tuning to the next channel is too long. Unfortunately, I can't fix this very easily, the spec only requires 300 us minimum gap, and the Galaxy S10e only uses a gap slightly longer than 300 us when the aux PHY is 1M or coded. The good news is that when the aux PHY is 2M (the most popular choice), at least the Galaxy S10e leaves over 1000 us separation between packets, so I have no timing issues.

With this said, I consider this issue resolved with commit cb7f1ff8a4135b974778bf4e6376afcb219b1b8d.

closed 2020-07-09 by sultanqasim

sultanqasim, 2020-07-09:
Reducing the auxPeriod introduced a regression of breaking connection establishment detection on aux channels, but I fixed this issue in a better way by rewording the channel hop scheduler logic in e1f737ec2d09e545849daf7ded7964e8218bd48c