nccgroup/Sniffle #22: pcap file timestamp are skipping ahead

sgawad, 2020-08-19:
I was trying to snif some packets between a custom device and a phone but the timestamp seems to skip ahead not sure why...
I attached the pcap. First jump at (12893) 33.86s  and then multiple other times. 
I see the same time jumps in the output. 

I also get the errors below:

C:\Users\S\Sniffle-master\python_cli>python sniff_receiver.py -eH -m 48:55:5c:00:02:16 -s COM10  -o datamotoz2.pcap > datamotoz2.txt
Ignoring message: Incorrect length field!
Traceback (most recent call last):
  File "C:\Users\S\Sniffle-master\python_cli\sniffle_hw.py", line 141, in recv_and_decode
    return PacketMessage(mbody, self.decoder_state)
  File "C:\Users\S\Sniffle-master\python_cli\sniffle_hw.py", line 244, in __init__
    raise SniffleHWPacketError("Incorrect length field!")
sniffle_hw.SniffleHWPacketError: Incorrect length field!
Ignoring message: Incorrect padding
Ignoring message: Incorrect padding
Ignoring message: Incorrect padding
Ignoring message: Invalid base64-encoded string: number of data characters (17) cannot be 1 more than a multiple of 4
Traceback (most recent call last):
  File "sniff_receiver.py", line 176, in <module>
    main()
  File "sniff_receiver.py", line 125, in main
    msg = hw.recv_and_decode()
  File "C:\Users\S\Sniffle-master\python_cli\sniffle_hw.py", line 138, in recv_and_decode
    mtype, mbody, pkt = self.recv_msg()
  File "C:\Users\S\Sniffle-master\python_cli\sniffle_hw.py", line 121, in recv_msg
    pkt = self.ser.readline()
  File "C:\Users\S\AppData\Local\Programs\Python\Python38-32\lib\site-packages\serial\serialwin32.py", line 286, in read
    result_ok = win32.GetOverlappedResult(
KeyboardInterrupt

[datamotoz2.zip](https://github.com/nccgroup/Sniffle/files/5096598/datamotoz2.zip)

Any help would be appreciated.

Thank you

sgawad, 2020-08-24:
@sultanqasim 
Tried it today using FTDI.. no more issues Thanks!

sultanqasim, 2020-08-24:
Great to hear!