nccgroup/Sniffle #23: Build Error -- rfc_ble5ScannerPar_t has no member named 'auxChRes'

StefanHri, 2020-09-30:
Hi,

unfortunately, I cannot build the firmware. I am getting the following errors: 

`stefan@t460s:~/Programs/sources/Sniffle/fw$ make

Generating configuration files...
Running script...
Validating...
Generating Code...
    '/ti/drivers/RF' -> '/ti/devices/driverlib'
    '/ti/drivers/RF' -> '/ti/drivers'
    '/ti/drivers' -> '/ti/devices/driverlib'
Unchanged /home/stefan/Programs/sources/Sniffle/fw/ti_devices_config.c...
Writing /home/stefan/Programs/sources/Sniffle/fw/ti_radio_config.c...
Unchanged /home/stefan/Programs/sources/Sniffle/fw/ti_radio_config.h...
Unchanged /home/stefan/Programs/sources/Sniffle/fw/ti_drivers_config.c...
Unchanged /home/stefan/Programs/sources/Sniffle/fw/ti_drivers_config.h...
Unchanged /home/stefan/Programs/sources/Sniffle/fw/ti_utils_build_linker.cmd.exp...
Unchanged /home/stefan/Programs/sources/Sniffle/fw/syscfg_c.rov.xs...
Building ti_radio_config.obj
ti_radio_config.c:484:6: error: 'rfc_ble5ScannerPar_t' {aka 'struct rfc_ble5ScannerPar_s'} has no member named 'auxChRes'
  484 |     .auxChRes = 0x00
      |      ^~~~~~~~
ti_radio_config.c:484:17: warning: excess elements in struct initializer
  484 |     .auxChRes = 0x00
      |                 ^~~~
ti_radio_config.c:484:17: note: (near initialization for 'ble5ScannerPar')
ti_radio_config.c:556:6: error: 'rfc_ble5InitiatorPar_t' {aka 'struct rfc_ble5InitiatorPar_s'} has no member named 'auxChRes'
  556 |     .auxChRes = 0x00
      |      ^~~~~~~~
ti_radio_config.c:556:17: warning: excess elements in struct initializer
  556 |     .auxChRes = 0x00
      |                 ^~~~
ti_radio_config.c:556:17: note: (near initialization for 'ble5InitiatorPar')
make: *** [makefile:142: ti_radio_config.obj] Error 1
`

Any ideas?

sultanqasim, 2020-09-30:
Which SDK version are you using? I've been building with the (now outdated) version 4.10.00.78 and might need to make minor updates to make newer SDKs work.

StefanHri, 2020-09-30:
I am using SDK version 4.20.01.04

sultanqasim, 2020-09-30:
OK. Try building with the older SDK version for now (4.10.00.78)

sultanqasim, 2020-10-20:
Fixed with e9f4cbd520c0fafdc7c6c50c012b76043785e0fc (see auxChRes.patch that needs to be applied to the SDK)

closed 2020-10-20 by sultanqasim