nccgroup/Sniffle #24: How to build the firmware for CC1352P1 Launchpad?

StefanHri, 2020-09-30:
Hi,
the readme mentions that it should be possible to build the firmware for CC1352P, however, some modifications are required. Can you detail those modifications since I have a CC1352P1 Launchpad that I want to use.

Thank you!
Stefan

sultanqasim, 2020-09-30:
I have not tried this, so I can't point you to exactly what has to be done. Mainly you'll need to create a directory like [this](https://github.com/nccgroup/Sniffle/tree/master/fw/CC1352) with the appropriate files from the SDK for the CC1352P1, edit the makefile to define a new platform variant for CC1352P1, and maybe add some code to configure the power amplifier.

sultanqasim, 2020-09-30:
If you do make the necessary modifications, feel free to submit a pull request

StefanHri, 2020-09-30:
OK sure. Can you point me an example code for the power amplifier

sultanqasim, 2020-09-30:
It looks like code to configure the antenna switch and power amplifier is auto-generated by SysConfig now, so you might not need to do anything for that. The linker file (.lds) and board specific init files (_fxns.c) come from SDK examples. The ccxml file is generated using Uniflash.

StefanHri, 2020-09-30:
Hi, 
I modified the make file and added a new folder CC1352P containing the files CC1352P1F3.ccxml, CC1352P1_LAUNCHXL_fxns.c, CC1352P1_LAUNCHXL_TIRTOS.lds. The make file is attached. 
[makefile.txt](https://github.com/nccgroup/Sniffle/files/5308529/makefile.txt)

Unfortunately, I get the following errors:

`
stefan@t460s:~/Programs/sources/Sniffle/fw$ make

Generating configuration files...
Running script...
Error: cannot set 'cmdList_ble' to cmdBle5GenericRx,cmdBle5Master,cmdBle5RadioSetup,cmdBle5Slave,cmdBleAdv,cmdBle5Initiator,cmdBle5Scanner,cmdFs: No option named cmdBle5RadioSetup defined, valid options are
cmdBle5RadioSetupPa
cmdFs
cmdTxTest
cmdRxTest
cmdBle5AdvAux
cmdBle5GenericRx
cmdNop
cmdRadioSetupPa
cmdFsOff
cmdSyncStopRat
cmdSyncStartRat
cmdResyncRat
cmdCount
cmdFsPowerup
cmdFsPowerdown
cmdSchImm
cmdCountBranch
cmdPatternCheck
cmdBleSlave
cmdBleMaster
cmdBleAdv
cmdBleAdvDir
cmdBleAdvNc
cmdBleAdvScan
cmdBleScanner
cmdBleInitiator
cmdBleGenericRx
cmdBleTxTest
cmdBle5Slave
cmdBle5Master
cmdBle5AdvExt
cmdBle5Scanner
cmdBle5Initiator
cmdBle5TxTest
cmdBle5Adv
cmdBle5AdvDir
cmdBle5AdvNc
cmdBle5AdvScan
    at Object.set (/opt/ti/sysconfig_1.4.0/dist/cli.js:26:288655)
    at eval (/home/stefan/Programs/sources/Sniffle/fw/sniffle.syscfg:47:54)
    at o.runAsUserScript (/opt/ti/sysconfig_1.4.0/dist/cli.js:26:249835)
    at Object.t.withDeprecatedAccess (/opt/ti/sysconfig_1.4.0/dist/cli.js:9:164743)
    at Object.t.runAsUserScript (/opt/ti/sysconfig_1.4.0/dist/cli.js:26:597250)
    at n.each.e (/opt/ti/sysconfig_1.4.0/dist/cli.js:26:249815)
    at Wt (/opt/ti/sysconfig_1.4.0/dist/cli.js:9:5239)
    at Function.Va (/opt/ti/sysconfig_1.4.0/dist/cli.js:9:40286)
    at Object.t.runScript (/opt/ti/sysconfig_1.4.0/dist/cli.js:26:249727)
    at <anonymous>
make: *** [makefile:149: syscfg] Error 1
`
 Somthing goes wrong with the syscfg but I dont know wat...

sultanqasim, 2020-10-01:
I'll look into this when I have some time

sultanqasim, 2020-10-20:
Ah, I see it needs `cmdBle5RadioSetupPa` instead of regular `cmdBle5RadioSetup` as used on models without the power amplifier. You can make it work by editing sniffle.syscfg to use the different setup command and modifying https://github.com/nccgroup/Sniffle/blob/master/fw/RadioWrapper.c#L74 to use `RF_cmdBle5RadioSetupPa` instead of `RF_cmdBle5RadioSetup`.

wudave, 2021-04-11:
Please advise details how to generate .lds, fxns.c and ccxml and where to place these files "The linker file (.lds) and board specific init files (_fxns.c) come from SDK examples. The ccxml file is generated using Uniflash.

Also, I used the makefile from StefanHri.  I also changed from "cmdBle5RadioSetup" to "cmdBle5RadioSetupPa" in sniffle.syscfg and from (RF_RadioSetup*)&RF_cmdBle5RadioSetup to (RF_RadioSetup*)&RF_cmdBle5RadioSetupPa in RadioWrapper.c.  But I was getting the error below.

~/Sniffle-master/fw$ make PLATFORM=CC1352P1F3
Generating configuration files...
Running script...
Validating...
Generating Code...
    '/ti/drivers/RF' -> '/ti/drivers'
.....
Building RadioWrapper.obj
RadioWrapper.c: In function 'RadioWrapper_init':
RadioWrapper.c:74:42: error: 'RF_cmdBle5RadioSetupPa' undeclared (first use in this function); did you mean 'RF_cmdBle5RadioSetup'?
   74 |                         (RF_RadioSetup*)&RF_cmdBle5RadioSetupPa, NULL);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~
      |                                          RF_cmdBle5RadioSetup
RadioWrapper.c:74:42: note: each undeclared identifier is reported only once for each function it appears in
make: *** [makefile:149: RadioWrapper.obj] Error 1

Can you please advise how to fix this issue?

sultanqasim, 2021-04-11:
You no longer need to add special lds and *fxns.c files. I don't have a CC1352P1F3 based launchpad to test with, but I could make a compatible branch for you to test if you'd like.

sultanqasim, 2021-04-11:
@wudave @StefanHri I made a branch for CC1352P1 launchpad that you can test: https://github.com/nccgroup/Sniffle/tree/cc1352p

Please let me know if it works for sniffing. It compiles fine, I just want to make sure it works and I'm not missing something with PA configuration.

wudave, 2021-04-11:
@sultanqasim Yes, it compiled and worked well.  Thanks for creating this tool and enhancing this capability.

sultanqasim, 2021-04-11:
https://github.com/nccgroup/Sniffle/commit/fe020045106d437c17444a15ed9a48d16bb4bf51

closed 2021-04-11 by sultanqasim