nccgroup/Sniffle #25: Minimum RSSI Value

tadegiac, 2020-12-06:
Hi there!

I am using the sniffer to compare the BLE 5.0 Longrange mode and the BLE 5.0 Legacy mode. 
The device I am using is a LAUNCHXL-CC26X2R1.

I couldn't receive any messages under a RSSI value of approximately  -90dBm. 
According to texas Instruments it should be possible to receive signals at -103dBm with the Longrange mode. 
Source: https://e2e.ti.com/blogs_/b/process/archive/2017/01/30/how-does-bluetooth-5-increase-the-achievable-range-of-a-bluetooth-low-energy-connection

Does anybody know whether this is possible? 
I was only able to find settings, that filter signals greater than a certain RSSI value.
Is there a setting where i can specify that i only want to reveice signals smaller than -90dBm or signals down to -103dBm?

Anyway, thanks for the great piece of software. It worked perfectly otherwise.

sultanqasim, 2020-12-06:
Hmm, I have not experimented with receiving really low RSSI frames. By default, sniff_receiver.py configured an RSSI floor of -80 for advertisements, but that can be overridden through the command line arguments. I did not deliberately set any other RSSI threshold, but it might be that some different radio tuning overrides are needed for hitting TI’s published RSSI floor. I did configure the hardware to ignore received frames with an invalid CRC, so maybe that’s contributing to loss of low RSSI data.

I can look into this more when I have time.

cschweers, 2020-12-07:
In my experience it is not possible to achieve such low receiving field strengths in a current home or office environment. People nowadays tend to use so much technology that uses the 2.4 GHz ISM band(WLAN, Zigbee ...) that there is a lot of noise on the RSSI floor.
Another problem will be that the LauchPad is normally supplied with power via the USB port of a PC or laptop.. This will be another source of noise that affects the sensitivity of the receiver.

closed 2021-07-21 by sultanqasim