nccgroup/Sniffle #27: Flashing Release using UniFlash 6.1.0 Debian and Win10

speedy-scooty, 2021-01-08:
When I verify Release 1.4 (sniffle_cc26x2r.out) before flashing to my Launchpad CC2652R1F (as auto-detected by UniFlash) I get the following errors, first a red pop-up window stating:

`Error! File: C:/Users/acer/Downloads/sniffle_cc26x2r.out: a data verification error occurred, file load failed.`

And in the UniFlash Console:
`
[1/8/2021, 11:17:26 PM] [INFO] Cortex_M3_0: GEL Output: Memory Map Initialization Complete.
[1/8/2021, 11:17:30 PM] [INFO] Cortex_M4_0: GEL Output: Memory Map Initialization Complete.
[1/8/2021, 11:17:32 PM] [ERROR] Cortex_M4_0: File Loader: Verification failed: Values at address 0x00000000 do not match Please verify target memory and memory map.
`

I only installed UniFlash, no sdk or anything and I think I am using the correct firmware (sniffle_cc26x2r.out) I think. Happens both on latest Debian and Windows. Any tips on how to go further?

Thank you!

sultanqasim, 2021-01-08:
`sniffle_cc26x2r.out` is the correct file for your board. I'm not sure why Uniflash is failing to write the code to flash though. Usually the physical connection is fine if Uniflash can detect the CC2652R1F.

Which version of Uniflash are you using? Is the XDS110 programmer firmware up to date? Uniflash will usually tell you if an update for the programmer is available. Also, try using Uniflash GUI to erase the flash, read out flash contents, try flashing Sniffle, then read out the contents again. Maybe that will give some insight.

dennisdebel, 2021-01-09:
Thank you for the quick reply. I Erased the flash in UniFlash GUI 6.1.0 (Settings & Utilities > Manual Erase > Erase Entire Flash), started a new session, read the target device memory (all FFFFFFFF's) and tried to verify the release image `sniffle_cc26x2r.out`  again. It gave the same error. Instead of bothering, I pressed Load Image (instead of Verify Image) and it loaded the firmware without a hitch it seemed! So, don't verify, LOAD!

sultanqasim, 2021-01-09:
Oh, that makes sense. Verify is to make sure the installed code matches the supplied image. Of course the code has to be loaded first :)

closed 2021-01-09 by sultanqasim