nccgroup/Sniffle #29: Support for CC2652R stick or CC2652RB stick

jarv-git, 2021-02-26:
Hi!

There are two sticks, one which is based on CC2652R (https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/) and another which is based on CC2652RB (https://slae.sh/projects/cc2652/), that its form factor is better than TI development boards.

Have you flashed your firmware in those devices or similar ones (I mean not TI development boards)? Do you think it will be hard to support those devices?

Thanks!

sultanqasim, 2021-02-26:
I have not tried either of these devices, but it should be trivial to make Sniffle firmware work on them. You might need to modify the UART PIN mapping if it's different on these boards compared to the Launchpad (haven't checked) but apart from that the Sniffle firmware should just work on them.

jarv-git, 2021-02-28:
Ok. Thanks.
I'll let you know if I get the firmware work on any of those devices.

closed 2021-03-01 by jarv-git