nccgroup/Sniffle #30: is CC2652P supported?

jurek4321, 2021-04-22:
Hello, i have cc2652p board. Is it supported? What i need to change to build for this?

sultanqasim, 2021-04-22:
Not currently but it should be simple to add support. You can see the commit adding CC1352P1 support for reference, but it might be easier for me to make the necessary changes than to explain how to make them. Might do it tonight if I have a moment.

sultanqasim, 2021-04-22:
@jurek4321 - does a standard TI Launchpad board for CC2652P exist? I can't find one offered by TI (only CC1352P1 and CC1352P4 versions). If it's not a standard off-the-shelf board you're using, I won't add support to the master branch but the changes needed to support it should be along the lines of my CC1352P1 support commit.

jurek4321, 2021-07-06:
Hey @sultanqasim. I read about porting to CC2652P and here https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/5.10.00.48/exports/docs/ble5stack/ble_user_guide/html/cc2652p/software-on-cc2652p.html i read "The CC2652P is pin compatible with, and functionally identical to the CC1352P device for 2.4 GHz operation. You can run CC1352P software examples on the CC2652P device without making any changes.". So i flashed using openocd command: openocd -f ocd.cfg -f /usr/local/share/openocd/scripts/target/ti_cc26x2.cfg -c "init" -c "reset halt" -c "flash write_image erase /home/user/files/sniffle/Sniffle/fw/builds/sniffle_cc1352p1.out". And it looks like its working on ebyte E72-2G4M20S1E. I run advertiser.py and i can detect beacon on my phone, but signal is a bit weak. CC2652P should have 20dbm, how i can increase tx power?

sultanqasim, 2021-07-06:
To adjust TX power, have a look at the `txPower` parameter of CMD_BLE5_RADIO_SETUP, and CMD_SET_TX_POWER in the reference manual for the CC13x2/26x2 family. I believe you need to set `txPower`in the radio setup command to 0xFFFF to enable the 20 dBm power amplifier.

jurek4321, 2021-07-07:
I tried to change it to 0xFFFF, but after that sniffle is not starting, not reacting to serial commands. I needed to comment " $(SYSCONFIG_TOOL) --board $(SYSCFG_BOARD) " from makefile and the last line from make clean, because it was overwriting this file. I also changed pin numbers for rf antenna switch in ti_drivers_config.h, it looks like ebyte module have it on different pins.
[image]

jurek4321, 2021-07-08:
@sultanqasim I found that i need to add pOverridesTx20 from smartrf studio to ti_radio_config.c, It is working but in smartrf studio there are options like enable or disable "whitening" and selection between "BLE, 1Mbps, BT4" or "BLE, 1Mbps, LE 1M", "Cap-array Tuning", "DC/DC Enable" What should i select for the best range?

sultanqasim, 2021-07-08:
You want BLE, 1Mbps, LE 1M with whitening enabled.

closed 2022-04-05 by sultanqasim