nccgroup/Sniffle #34: is the ZZH cc2652r stick supported?

maichai, 2021-07-07:
The ZZH cc2652r stick would be a better format for pentesting in the field. Would you be willing to support that in the future with a release image?

[zzh stick on tindie](https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/)

sultanqasim, 2021-07-07:
Looks like a convenient device. Not an immediate priority for me, but it shouldn't be too difficult to add support for it. I could look into this at some point, or I'd be happy to accept patches adding support for it.

sultanqasim, 2024-04-22:
Still seems to be out of stock / out of production. I do want to add support for some such sticks though, and the existing firmware already works on the ZigStar Stick V4 (as mentioned in https://github.com/nccgroup/Sniffle/issues/35#issuecomment-944904407).

closed 2024-04-22 by sultanqasim