nccgroup/Sniffle #36: Errors with 1.5 on CC1352-P2

maichai, 2021-07-20:
I use uniflash to load the 1.5 release file sniffle_cc1352p1.out onto a CC1352-P2 Launchpad. Before that I completely erased the flash using uniflash. After loading the file I verified it successfully.

I see both serial devices as both tty and cu under macOS. The red LED is flashing. The green LED is on.

However, running the python scripts using the first serial gives errors:

-always in the beginning, once: `Ignoring message: Incorrect padding`
-many of these: `Ignoring message due to missing CRLF`
-some of these: `sniffle_hw.py", line 221, in recv_and_decode; raise SniffleHWPacketError("Unknown message type 0x%02X!" % mtype); sniffle_hw.SniffleHWPacketError: Unknown message type 0x<various>!`

No devices are found by scanner.py or sniff_receiver.py
I also tried reset.py and before running scanner or sniff receiver, but it did not solve the problem.

I would love to test your sniffle tool, but it does not seem to work on CC1352-P2

sultanqasim, 2021-07-20:
I use the cu.usb… serial device on Mac. From your error messages, it sounds like either an issue in UART communications with your computer, or a mismatch between the firmware version and host side Python software version. If the red LED blinks when the board is powered up and nearby BLE devices are advertising, then the firmware is working.

I’ll probably be tagging the latest git revision as version 1.6 today after some more testing. You can try the prebuilt firmware binaries for version 1.6 together with the corresponding host side Python scripts.

sultanqasim, 2021-07-20:
Just to be clear, version 1.5 firmware is not compatible with the current master revision of host side Python code. Master revision host side code needs master revision firmware to work. When using prebuilt tagged firmware versions, use it with host side Python code corresponding to that tag.

maichai, 2021-07-20:
OK I flashed the v1.6 FW and now it works with the cli tools. This is definitely something you should write promimently in the README. It s absolutely non-obvious, I read read the README carefully because the stuff didn't work, but there was no such hint.

sultanqasim, 2021-07-21:
Readme says “When using prebuilt firmware, be sure to use the Python code corresponding to the release tag rather than master to avoid compatibility issues with firmware that is behind the master branch.”

closed 2021-07-23 by sultanqasim