nccgroup/Sniffle #38: Feature request: integragtion with Wireshark

maichai, 2021-07-20:
I have seen that the sniffer can write pcaps and they work in wireshark. Would it be possible to have a module in wireshark that speaks with the sniffle firmware directly so we can directly sniff in wireshark?

sultanqasim, 2021-07-21:
Anything is possible, would just take time to implement 😉

piotrwiniarczyk-silvair, 2021-09-07:
The documentation for this is here (extcap):
https://www.wireshark.org/docs/wsdg_html_chunked/ChCaptureExtcap.html
and a working example can be seen in Nordic sniffer here:
https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le/download#infotabs
with installation documentation here:
https://infocenter.nordicsemi.com/topic/ug_sniffer_ble/UG/sniffer_ble/installing_sniffer_plugin.html

jslawek, 2021-09-29:
I agree, extcap would be nice. The nRF toolbar in Wireshark is also very convenient by the way. It does not look too complex to implement something similar, maybe even starting with their extcap - which is written in python, and with (looks like) permissive license: 
> without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is furnished to do so.
 
But for starters maybe even easier would be just to write the packets to a FIFO pipe instead of pcap file, and then open the pipe in wireshark?
Just like in [btlejack](https://github.com/virtualabs/btlejack#dumping-live-packets-with-wireshark): you specify a FIFO file as a parameter, and then issue `wireshark -i PIPEFILE` to get the live packets directly.

It is implemented in [btlejack's python cli](https://github.com/virtualabs/btlejack/blob/master/btlejack/pcap.py), so it should be easy to port.

jslawek, 2021-09-29:
Ohh... I just missed the fact that extcap is already there: https://github.com/nccgroup/Sniffle/pull/42. I will test it and maybe look into possibilities for adding a tolbar similar to the nRF one.

homewsn, 2021-10-21:
I wrote [bsniffhub](https://github.com/homewsn/bsniffhub) exactly for this purpose. It interfaces Sniffle firmware directly to Wireshark, has both a console version and a GUI version, and can be used for both Linux and Windows.

maichai, 2021-12-23:
That is awesome! I will try that.

sultanqasim, 2022-08-19:
Added in v1.7

closed 2022-08-19 by sultanqasim