nccgroup/Sniffle #43: Unable to build working firmware

mh-, 2021-09-17:
Hi, 
First of all thanks a lot for publishing this excellent work!

I can successfully flash release v1.6 (sniffle_cc2652rb.out) to my LP-CC2652RB. The red LED blinks, and the python cli tools work as intended.

But when I try to build this binary myself, it doesn't work.
I do `make clean` and `make platform=CC2652RB1F load`, and the process generates a `sniffle.out` file, but when I flash that, the red LED stays off and there's no serial output from the board.
I tried this on macOS and Ubuntu, using the exact versions of the ARM compiler and the TI SDK that you mentioned, but both sniffle.out files don't work. I also checked out the v1.6 tag in git, to use the same version that you released, but to no avail.

Is there something special you do additionally, in order to create sniffle_cc2652rb.out?
Or any idea on how to debug this?

Thanks
Michael

sultanqasim, 2021-09-17:
Hmm, strange. When I build a release, all I do is run this script: https://github.com/nccgroup/Sniffle/blob/master/fw/build_all_platforms.sh

This is the exact compiler I'm using:

```
sultan@sultan-neon-vm:~/VMShared/sniffle/fw$ ~/arm_tools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I suppose you could try diffing the generated .out elf files

mh-, 2021-09-17:
Yes, same compiler:
```
% ~/arm_tools/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc --version

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

Built by me:
```
% ls -l builds                                                                 
-rwxr-xr-x  1 mh  staff  1802704 Sep 17 19:02 sniffle_cc2652rb.out
```

Built by you:
```
% ls -l sniffle_cc2652rb-1.6.out 
-rw-r--r--@ 1 mh  staff  1802496 Sep 16 17:17 sniffle_cc2652rb-1.6.out 
```

Of course this yields: `Binary files sniffle_cc2652rb-1.6.out and builds/sniffle_cc2652rb.out differ`

mh-, 2021-09-17:
Ok...
This builds the wrong platform:
```
% make platform=CC2652RB1F
```

But this builds the correct _PLATFORM_:
```
% make PLATFORM=CC2652RB1F
```

closed 2021-09-17 by mh-