nccgroup/Sniffle #44: Release LL relay features

mdlam92, 2022-05-17:
Don't gatekeep, release it to the public. We want to test!

mdxs, 2024-04-24:
See https://github.com/nccgroup/Sniffle?tab=readme-ov-file#transmit-functionality

XenoKovah, 2025-01-22:
@sultanqasim I just noticed that WHAD claimed relay functionality in their Defcon talk, so perhaps you could close tickets like this (and #85) by just saying "We're not going to release the relay, but you can try with WHAD."

sultanqasim, 2025-01-22:
I don't want to say we're not going to release it because I do want to release it, though I need authorization from our management to allow that. That said, since both WHAD and some [Chinese researchers](https://conference.hitb.org/hitbsecconf2023hkt/materials/D2%20COMMSEC%20-%20The%20Tragedy%20of%20Bluetooth%20Low%20Energy%20-%20Linfeng%20Xiao%20&%20Dalin%20Yang.pdf) seem to have recreated the missing host-side bits for the link layer relay attack with Sniffle, and the WHAD team have made theirs public, it may make it easier to justify releasing it.

sultanqasim, 2025-07-03:
Released :)

closed 2025-07-03 by sultanqasim