nccgroup/Sniffle #45: Successful build of Sniffle for CC2651P3?

stevewillson, 2022-06-03:
The LP-CC2651P3 is listed as a supported hardware platform for Sniffle. I attempt to build Sniffle by following the instructions on the `README.md` file. But after flashing the built firmware to the LP-CC2651P3 board, I am not able to observe any output when running `scanner.py` or `sniff_receiver.py`

Build procedure:
1. Modify `makefile`, remove the dependency on syscfg
BEFORE: `%.obj: %.c $(CONFIGPKG)/compiler.opt syscfg`
AFTER: `%.obj: %.c $(CONFIGPKG)/compiler.opt`
2. Run `make syscfg`, specify `PLATFORM = CC2651P31`
3. Open the file `ti_radio_config.c` and comment out the line:
`.extFilterConfig.bAcceptSyncInfo = 0x0` according to: https://e2e.ti.com/support/wireless-connectivity/bluetooth-group/bluetooth/f/bluetooth-forum/1090449/cc2651p3-sysconfig-including-bacceptsyncinfo-parameter-for-cmd_ble5_scanner
4. Run `make sniffle.out` to build the `sniffle.out` firmware binary.
5. Flash the file sniffle.out to the LP-CC2651P3 using `make load`

I receive a "Success" message after running `make load`

I run some of the python_cli tools `scanner.py` or `sniff_receiver.py` but they do not output any information.

I have multiple BLE devices near the LP-CC2651P3 board so I believe that I should see some output when the above tools are run.

Version Information:

DSLite version 12.0.0.2728
ARM GCC Toolchain: gcc version 11.2.1 20220111 (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 
SimpleLink SDK: simplelink_cc13xx_cc26xx_sdk_6_10_00_29
Hardware: LP-CC2651P3

Information about the `sniffle.out` file:

$ sha1sum sniffle.out 
ffb94008161f67ab48b89829d16e4da806111760  sniffle.out
$ file sniffle.out 
sniffle.out: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped

$ ls -la sniffle.out 
File size: 1579852 bytes

Any help in debugging this issue is greatly appreciated.

sultanqasim, 2022-06-03:
The procedure you described sounds correct, so I’m not yet sure why it isn’t working for you. I’m away from my CC2651 board at the moment, but I have successfully tested it on the CC2651P3 (at least with SDK 5.40). I’ll make a build myself shortly and let you try it.

sultanqasim, 2022-06-03:
Here's a build I made following largely the same steps you did. Unzip it, and flash it with Uniflash.
[sniffle_cc2651p3.out.zip](https://github.com/nccgroup/Sniffle/files/8829530/sniffle_cc2651p3.out.zip)

By chance, did you forget to specify `PLATFORM = CC2651P31` when you invoked `make sniffle.out` and/or `make load`?

stevewillson, 2022-06-03:
Thank you! I loaded the [sniffle_cc2651p3.out.zip](https://github.com/nccgroup/Sniffle/files/8829530/sniffle_cc2651p3.out.zip) and ran the `scanner.py` tool. It immediately showed output and appears to work.

I noticed the file that you provided is 2.2MB in size while the file I built is 1.6MB. 

I attempted to isolate the problem by removing all references to other platforms from the `makefile`, but still ended up with a `sniffle.out` file size of 1.6MB that did not work.

I reverted to the original `makefile` in the master branch in case I inadvertently messed something up, then followed the steps I mentioned in the earlier post to build the file. I specified the platform with `$ export PLATFORM=CC2651P31` and made sure that it was set prior to running `make syscfg`, `make sniffle.out`, and `make load`, the `sniffle.out` file was still 1.6MB and did not work when loaded to the CC2651P3 board.

I'm interested in building Sniffle, are there other recommendations to check?

sultanqasim, 2022-06-03:
How big is the firmware (.out ELF file) if you run `make clean` and then `make CC2652R1F` with an unmodified makefile? It should be around 2.2 MB as well. The hack around commenting out the sysconfig invocation (for bAcceptSyncInfo) is only needed for the CC2651 family.

The procedure I used to build for the CC2651P3 was:

1. Start with an unmodified makefile, and run `make clean`
2. Run `make PLATFORM=CC2651P31` and watch the build proceed till it fails with the `bAcceptSyncInfo` error.
3. Comment out line 172 of the makefile (https://github.com/nccgroup/Sniffle/blob/master/fw/makefile#L172) and remove the `bAcceptSyncInfo` line from `ti_radio_config.c`.
4. Run `make PLATFORM=CC2651P31` again

stevewillson, 2022-06-03:
When using `$ make PLATFORM=CC2652R1F` the `sniffle.out` firmware file is 1.6MB (as reported by `ls -lh sniffle.out`). 

I will check that I set up the environment correctly.

Steps:
`$ git checkout makefile`
`$ make clean`
`$ make PLATFORM=CC2652R1F`
`$ ls -lh sniffle.out`

The `sniffle.out` file size is 1.6MB.

$ sha1sum sniffle.out (for the CC2652R1F platform)
9bfd562dda461cd1f5a316f83d3b7d99a7dd5bd5  sniffle.out

I'm unable to verify if the file works correctly because I only have a CC2651P3 Development Kit.

I'm building this on an Ubuntu 22.04 OS machine.

sultanqasim, 2022-06-03:
Which compiler are you using? Use a compiler from this series: https://developer.arm.com/downloads/-/gnu-rm

(they're now deprecated, but I had issues with their replacement)

I've been using `gcc-arm-none-eabi-9-2019-q4-major` from there, but their last 10.3 release should also work.

stevewillson, 2022-06-03:
The `gcc-arm-none-eabi-9-2019-q4-major` compiler works, thank you!

I used the 9-2019-q4-major release.

I downloaded the compiler from: https://developer.arm.com/downloads/-/gnu-rm

The specific version I am using
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 

I also tried the 10.3 release, but it did not work for me. At first, I used the 11.2 release (which also didn't work).

Below are the versions I tried in the `imports.mak` file (`$(HOME)/ti/simplelink_cc13xx_cc26xx_sdk_6_10_00_29/imports.mak`)

```
GCC_ARMCOMPILER         ?= /home/user/arm_tools/gcc-arm-none-eabi-9-2019-q4-major # WORKS
#GCC_ARMCOMPILER        ?= /home/user/arm_tools/gcc-arm-11.2-2022.02-x86_64-arm-none-eabi # DOESN'T WORK
#GCC_ARMCOMPILER        ?= /home/user/arm_tools/gcc-arm-none-eabi-10.3-2021.10 # DOESN'T WORK
```

closed 2022-06-03 by stevewillson