nccgroup/Sniffle #50: Forcing a repairing of bluetooth devices

PowerPress, 2022-10-07:
Can this or any other software you know of send a LMP message that it lost the pairing secret to force a repairing to other bluetooth devices you are trying to sniff?

sultanqasim, 2024-04-04:
Sniffle actually does support transmitting (as a an advertiser, initiator, connection master, or connection slave), though I haven't documented that functionality. The relay attack research I published a while back was built on Sniffle's mostly undocumented transmit capabilities. While BLE doesn't use LMP, you can send link layer PDUs to reject connection startup, and then send SMP (Security Manager Protocol) messages to encourage re-pairing.

sultanqasim, 2024-04-16:
Closing as this isn't an "issue" and LMP is not relevant to BLE.

closed 2024-04-16 by sultanqasim