nccgroup/Sniffle #52: Capture with Wireshark using Sniffle plugin freezes on Linux

mdxs, 2023-01-30:
When I try to capture with Wireshark using the Sniffle 1.7 plugin, it freezes on my Linux setup.

When I use the command-line tools, I can capture into a PCAP that I can load into Wireshark just fine.

Perhaps this is specific to my environment, using:

- Xubuntu 22.04.1 LTS (as a VMware 16.2.4 Guest inside a Windows 11 host)
- Wireshark 3.6.2 (Git v3.6.2 packaged as 3.6.2-2)
- Sniffle 1.7 expanded from .zip (from the release Version 1.7)
- LAUNCHXL-CC26X2R1 with `sniffle_cc2652r1.out` (from the release Version 1.7)
- Using the instructions from the README.md (tried both the files copied and the symlink approach)

Strange is also that (after first use after install) using the "Sniffle BLE sniffer: sniffle" interface, I recall Wireshark showing a dialog to configure the Sniffle settings. However it doesn't seem to allow me to get into those settings again (though I did see them reflected in `~/.config/wireshark/preferences` (for example to select the serial port to be used).

sultanqasim, 2023-01-30:
When you say it "freezes", do you mean the Wireshark program freezes, or it just fails to capture anything?

mdxs, 2023-01-30:
Both: wireshark freezes and it appears not to capture anything in the process.

I'll clean my virtual machine and replay my steps to provide some more details of what is happening.

mdxs, 2023-01-30:
I've re-installed Wireshark and ensured that the previous `~/.config/wireshark/` was removed in between.

I then tested that Wireshark could load a Sniffle captured PCAP just fine.

Then I used the following to install the Sniflle plugin for Wireshark (and checked that `sniffle_extcap.py` is executable):
```
mkdir -p ~/.config/wireshark/extcap/
ln -s ~/Sniffle/python_cli/sniffle_extcap.py ~/.config/wireshark/extcap
```

Then started Wireshark from the command-line with `--log-level info` to see if I could spot some errors. The "Sniffle BLE sniffer: sniffle" interface shows under the Capture options. When I start the Capture with `sniffle` in this fresh mode, it shows:

[image]

So far, so good; except that the Sniffer serial port needs to be changed AFAIK; and I wanted to capture the communications of a specific BLE device using MAC filter; so I changed those in the dialog:

[image]

After clicking on Start, it freezes showing the following (the console shows the `info` log):

[image]

Trying to close the Wireshark application shows:

[image]

mdxs, 2023-01-30:
With `debug` log level, the console from which Wireshark is started shows:

[image]

sultanqasim, 2023-01-30:
Thanks for the detailed report, I will look into this.

mdxs, 2023-01-30:
Ah, and when the first freeze has been seen: the dialog to configure the Sniffle capture isn't shown. It tries to start a capture right away. As the `debug` console output shows, it appears to take the inputs from the previous run. Perhaps this is due to me not being that familiar with Wireshark... maybe this behavior is expected and I don't know how to re-enter the Sniffle dialog.

sultanqasim, 2023-01-30:
After the first time the sniffer is used, it saves the settings from the first run. To change the settings on subsequent runs, click the gear icon to the left of the Sniffle capture device on the Wireshark start screen.

sultanqasim, 2023-01-30:
At least on KDE Neon (based on Ubuntu 22.04) with Wireshark 3.6.2, capture is working fine for me. All I did to install was:

```
mkdir -p ~/.config/wireshark/extcap/
ln -s $(pwd)/sniffle_extcap.py ~/.config/wireshark/extcap
```

I then launched Wireshark, clicked the gear icon beside Sniffle to bring up capture settings, picked the right serial port, and captured. It worked fine, and adding a MAC filter also worked fine for me.

I'm still looking into what could be going wrong for your setup.

sultanqasim, 2023-01-30:
You can also try setting the SNIFFLE_LOG_FILE environment variable before launching Wireshark to capture extra logs (see https://github.com/nccgroup/Sniffle/blob/master/python_cli/sniffle_extcap.py#L79)

RanitPradhan, 2023-04-18:
> You can also try setting the SNIFFLE_LOG_FILE environment variable before launching Wireshark to capture extra logs (see https://github.com/nccgroup/Sniffle/blob/master/python_cli/sniffle_extcap.py#L79)

Is it worked @mdxs ?

sultanqasim, 2024-04-03:
@jsmif did you try with /dev/ttyACM0 instead of /dev/ttyACM1? Usually ttyACM0 is the Sniffle serial port, and ttyACM1 is the XDS110 debugger control port.

sultanqasim, 2024-04-03:
Oh, that certainly helps narrow down the cause. What are the udev rules you have set up for the Launchpad's USB UART? Can your user account access the tty (without sudo)? I have udev rules set up to allow accessing the tty as my regular user account.

Perhaps I should update the documentation to make this clear, and add some error handling and reporting to the Wireshark plugin in case of permission errors.

sultanqasim, 2024-04-03:
> I don't know what udev rules are or how to set them, so the VM is just default. So no, I can't access the serial port without sudo. So yes, if there's some commands that you ran to allow for serial access from the user by default, that's probably it, and adding those to the instructions would be appreciated. (And I'll give them a try once available even though I have the workaround now.)

Ah, well that explains it. Udev rules (typically located in `/etc/udev/rules.d/`) can be used to specify file permissions for access to USB devices, among other things. Typically you specify the USB VID, PID, permissions, and associated group (for example `plugdev`). People often write their own custom udev rules, or use rules defined by the device manufacturer. I'll share the udev rules I use in a bit.

sultanqasim, 2024-04-05:
For reference, these are the udev rules written by TI that are installed with UniFlash at `/etc/udev/rules.d/71-ti-permissions.rules`:

~~~
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="a6d0",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="a6d1",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="6010",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="bcd9",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="bcda",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1cbe",ATTRS{idProduct}=="00fd",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1cbe",ATTRS{idProduct}=="00ff",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="bef1",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="bef2",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="bef3",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="bef4",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="c32a",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="0451",ATTRS{idProduct}=="c31a",MODE:="0666"
ATTRS{idVendor}=="0451",ATTRS{idProduct}=="bef0",ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="0c55",ATTRS{idProduct}=="0220",ENV{ID_MM_DEVICE_IGNORE}="1"
KERNEL=="ttyACM[0-9]*",MODE:="0666"
~~~

mdxs, 2024-04-24:
Thanks for picking this up, I've created a new VM to confirm in my environment (different from original post):

- Xubuntu 22.04.4 LTS (as a VMware 17.5.1 Guest inside a Windows 11 host)
- Wireshark 3.6.2 (Git v3.6.2 packaged as 3.6.2-2)
- Sniffle 1.8 expanded from .zip (from the release Version 1.8)
- LAUNCHXL-CC26X2R1 with `sniffle_cc2652r1.out` (from the release Version 1.8)
- LAUNCHXL-CC26X2R1 with `firmware_3.0.0.28_fastuart.bin` applied
- Using the instructions from the README.md with the symlink approach

In that initial setup, starting Wireshark with the capture plugin showed the "Couldn't run /usr/bin/dumpcap in child process. Permission denied" error dialog (same as @jsmif showed in https://github.com/nccgroup/Sniffle/issues/52#issuecomment-2064515638).

Fixed this with: `sudo chmod +x /usr/bin/dumpcap` and Wireshark happily started to capture the packets.

So I can confirm that the issue has been resolved in Sniffle 1.8 as long as the user sets the permissions on `dumpcap` correctly, which might be environment specific. Perhaps that can be added to the README.md as a hint.

closed 2024-04-24 by mdxs