nccgroup/Sniffle #54: update to use UART2 drivers

stevewillson, 2023-04-20:
SimpleLink™ cc13xx_cc26xx SDK 6.20.00.29 removes support for UART in favor of the UART2 driver.

https://software-dl.ti.com/simplelink/esd/simplelink_cc13xx_cc26xx_sdk/6.20.00.29/exports/release_notes_simplelink_cc13xx_cc26xx_sdk_6_20_00_29.html

Sniffle appears to use UART in messenger.c/h files and also the sniffle.syscfg file.

UART is removed from the SimpleLink SDK in 6.20.00.29. Sniffle will not build when using SimpleLInk SDK 6.20.00.29 or newer

sultanqasim, 2023-04-21:
I investigated this, though on SDK 6.20.00.29 at least, under high throughput and relay attacks scenarios, I encountered data loss and stalling bugs in the UART2 driver that were not happening with the old UART driver. From what I recall, the old driver was also still present in SDK 6.20 despite the release notes saying it was removed, so I kept using it in the interim. At some point I’ll need to test UART2 on the latest SDK to see if the bugs have been fixed, and if not, l’ll either have to debug UART2 and send a fix to TI, or ship my own copy of the old UART driver with Sniffle.

stevewillson, 2023-04-21:
Thank you, I recently downloaded SDK 6.41.00.17 for Linux and it appears that the UART drivers are removed from the `ti/drivers/` directory. I was incorrect to say that SDK 6.20.00.29 removed the UART driver, I updated my first comment with that change.

sultanqasim, 2024-04-07:
6813060490a830ca7a289b4887385c277baf1470

closed 2024-04-07 by sultanqasim