nccgroup/Sniffle #58: Sending UART commands from external MCU

Michaelszeng, 2023-05-27:
Hi,

How could I go about sending UART commands to my dev kit from an external MCU?

I've looked through the python_cli code, and understand all the UART commands that need to be sent. I also see that that TI Launchpad has TX and RX pins broken out (pins DIO2 and DIO3 on the LP-CC2652RB). What would I need to change in the firmware to be able to use UART through those pins?

In addition, would a lower baud rate like 115200 still work (instead of the 2000000 that it is currently programmed to use)?

sultanqasim, 2023-05-27:
This is absolutely possible and should be quite easy. For the UART pin reassignment, you’ll need to edit the SysConfig file: https://github.com/nccgroup/Sniffle/blob/ed55da354371c8b850f5e18780c77f3e1efad816/fw/sniffle.syscfg#L96

You can use the TI SysConfig utility to generate the correct lines to put there for configuring the UART with the desired pins. Alternatively, if you just want to use the UART pins used by the debugger, just pull the debug UART jumpers on the Launchpad board and hook up your own wires.

For baud rate, you can absolutely do 115200 baud, just edit messenger.c accordingly and recompile: https://github.com/nccgroup/Sniffle/blob/ed55da354371c8b850f5e18780c77f3e1efad816/fw/messenger.c#L20

Be aware that the Sniffle firmware will only compile with the exact (outdated) SDK version specified in the README. Using any other SDK version to compile will encounter issues. At some point, I’ll need to make time to migrate to the latest 7.10 SDK, though it’s an ongoing task as the TI SDK evolves.

mdxs, 2024-04-24:
I think this ticket can be closed as answered.

sultanqasim, 2024-04-26:
Closing as answered

closed 2024-04-26 by sultanqasim