nccgroup/Sniffle #59: Blank Output after Build for CC2652RB1F

Michaelszeng, 2023-05-28:
I've seen issue #45 and am having basically an identical issue. I'm using a CC2652RB launchpad, and a windows 10 laptop.

I've run **_make clean_**, then **_make PLATFORM=CC2652RB1F_**, then **_make load,_** which all are successful. However, when I run **_python sniff_receiver.py_**, nothing is printed to the terminal. Likewise for running **_python scanner.py_**, it does not find any Scan Results. Both of these programs also will not terminate until I unplug the launchpad. 

My sniffle.out file is also 605 KB, a bit smaller file size than the pre-built .out file (621 KB).

Flashing the prebuilt binaries in the releases section works, but I would like to build myself so that I can modify the code in the future (see my other issue #58 ).

I've tried the fixes offered in issue #45, namely building with the gcc-arm-none-eabi-9-2019-q4-major compiler (I've made sure to set the correct path to it in the imports.mak file). The only modification I've made to the makefile is the SIMPLELINK_SDK_INSTALL_DIR. Any suggestions?

sultanqasim, 2023-05-28:
Try arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi as mentioned in the readme.

Michaelszeng, 2023-05-28:
Thanks--I've tried that version as well before switching to gcc-arm-none-eabi-9-2019-q4-major, same issue though unfortunately.

sultanqasim, 2023-05-28:
Hmm, you can try diffing the binaries. The 11.3 compiler I mentioned is the exact version I’ve been using to build recent versions of the firmware.

sultanqasim, 2023-05-28:
Oh, by the way, you need to specify the PLATFORM= value every time you invoke make, including load (and even clean to be safe).

Michaelszeng, 2023-05-28:
Thanks--I actually wasn't specifying `PLATFORM=` every time I invoked make. Just tried again, but same result unfortunately.

I've thought about this a while and am not sure what else I could be doing wrong; I've also downloaded a software to try comparing binaries but I'm not too sure how this helps?

Could I ask a big favor from you-- could you simply build the binary? The only 2 changes I would like to make to the firmware is to change the baud rate in messenger.c to 115200 and the UART port that it uses. To change the UART port, I think changing this line:

https://github.com/nccgroup/Sniffle/blob/master/fw/sniffle.syscfg#L96

to this:

```
UART1.uart.txPin.$assign  = "boosterpack.30";
UART1.uart.rxPin.$assign  = "boosterpack.29";
UART1.txPinInstance.$name = "CONFIG_GPIO_0";
UART1.rxPinInstance.$name = "CONFIG_GPIO_1";
```

(assigns the tx pin to DIO1 and the RX pin to DIO0), should work (although I'm not the most experience in sysconfig).

I am asking a lot but I would really appreciate the help--I'm really passionate about this project (has to do with detect and avoid for drones) and Sniffle has been incredibly helpful already.

elec-tra, 2023-07-27:
Hi @Michaelszeng I tried to build for the device CC2652RB1F, for me, it is working fine. Can you provide more information about your build environment?

sultanqasim, 2024-04-26:
@Michaelszeng Do you still need assistance with this?

sultanqasim, 2024-05-07:
Closing due to inactivity and inability to reproduce the issue