nccgroup/Sniffle #61: Does Sniffle support active scanning mode?

0dayAK, 2023-09-16:
I found during testing Sniffle that the cmd_scan() command in scanner.py can only perform passive scanning, meaning that the Sniffle firmware does not send out SCAN_REQ packets (it's not in active scanning mode). Assuming there are no BLE clients around, then the scanner.py in Sniffle cannot scan any nearby BLE servers. Is it that the Sniffle firmware has not implemented the active scanning feature, or am I using it incorrectly?

Any help much appreciated!

sultanqasim, 2023-09-16:
The scan command should perform active scanning. If it isn’t sending scan requests, something is wrong.

sultanqasim, 2023-09-16:
The scanner.py script should be a simple way of trying out this feature. The scan requests will not be shown in the PCAP (the firmware doesn’t send to the host the scan requests that it’s sending), but they absolutely should be getting sent by the firmware. The firmware will only send the host the advertisements and scan responses.

0dayAK, 2023-09-18:
@sultanqasim Thank you for your patient answers. I finally realized that there was a mistake in my code when I implemented Sniffle in C language. Thank you again for your help!

closed 2023-09-18 by 0dayAK