nccgroup/Sniffle #63: I'd like to apply it to extend the Bluetooth range.

76donghyun, 2023-10-18:
I was deeply impressed after watching your presentation on YouTube and decided to visit your GitHub. 

I want to try applying your attack method to connect a smartphone and AirPods. 

Is it possible to connect a smartphone and AirPods at a distance greater than what's typically achievable with regular Bluetooth using your attack method?

If it's possible, can I simply use the method described on GitHub? Or does GitHub only have methods for attacking Tesla? 

Thank you.

sultanqasim, 2023-10-18:
The relay attack code can also be used for ordinary range extension. Sniffle only supports Bluetooth LE however, and current generation AirPods use BT Classic I believe, though LE audio is an up-and-coming standard.

76donghyun, 2023-10-18:
Does that mean I can use the Launchpad Board you mentioned when attacking other LE?

sultanqasim, 2023-10-18:
Yes, it can be used as a general purpose BLE range extender.

76donghyun, 2023-11-03:
root@WIN-8O1QD981DMQ:~/ti/Sniffle/fw# make load
Error: unknown property: ringBufferSize
    at cb (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:104:13)
    at Object.withDeprecatedAccess (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/deprecatedAccessGuard.ts:14:10)
    at Object.runAsUserScript (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/scriptingGuard.ts:49:3)
    at iteratee (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:100:5)
    at baseEach (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:530:11)
    at Function.each (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:9409:52)
    at iteratee (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:98:6)
    at baseEach (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:530:11)
    at Function.each (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:9409:52)
    at Object.t.runScripts (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:86:4)
Error: unknown property: ringBufferSize
    at cb (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:104:13)
    at Object.withDeprecatedAccess (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/deprecatedAccessGuard.ts:14:10)
    at Object.runAsUserScript (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/scriptingGuard.ts:49:3)
    at iteratee (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:100:5)
    at baseEach (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:530:11)
    at Function.each (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:9409:52)
    at iteratee (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:98:6)
    at baseEach (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:530:11)
    at Function.each (/root/ti/sysconfig_1.13.0/dist/webpack:/node_modules/lodash/lodash.js:9409:52)
    at Object.t.runScripts (/root/ti/sysconfig_1.13.0/dist/webpack:/src/pinmux/services/scripting/runScript.ts:86:4)
Building DelayHopTrigger.obj
DelayHopTrigger.c:14:10: fatal error: ti_drivers_config.h: No such file or directory
   14 | #include "ti_drivers_config.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [makefile:186: DelayHopTrigger.obj] Error 1

I'm getting the following compliation error. How can I resolve it?

mdxs, 2024-04-24:
I think the original question has been answered, and thus, this ticket can be closed.

The compilation error reported later on is not relevant to the original question.

sultanqasim, 2024-05-05:
@76donghyun your build error sounds like you're using the wrong SDK version. Try with the latest firmware and the exact SDK version specified in the readme.

closed 2024-06-05 by sultanqasim