nccgroup/Sniffle #66: Coded Phy Advertising Receprion.

froloffw7, 2024-04-06:
Hello, 

Does this BLE scanner have the capability to receive Coded PHY advertising?
I rebuild your project for the CC2652P based [E72-2G4M20S1E](https://www.ebyte.com/en/product-view-news.html?id=1002) module from EBYTE.
The only changes, I made were:
- switched RF Control Pins to DIO05/DIO06, according to the module schematic
- decreased UART speed t to 1000000 baud
- optionally activate UART hardware flow control.

Unfortunately, the scanner is only able to receive and display 1M PHY advertising and never Coded PHY. I have a couple of devices transmitting advertising in Coded PHY mode and can easily find them with nRF Connect scanner from Nordic.

Command line I use to run scanner:
`./sniff_receiver.py -le`

Does it supposedly continue to receive and display 1M PHY data despite the -le flags?

sultanqasim, 2024-04-06:
Your command `./sniff_receiver.py -le` looks correct for receiving advertisements on the coded PHY. With that command, it should only receive advertisements (on primary advertising channels) using the long range coded PHY, not the 1M PHY. Those advertisements will contain an AuxPtr that can point to an auxiliary advertisement on any channel and PHY. For example, I just tried it right now with my phone putting out coded advertisements, with auxiliary advertisements on the 2M PHY, and it worked fine.

~~~
Timestamp: 12.866209    Length: 9       RSSI: -59       Channel: 37     PHY: Coded (S=2)
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 8 PHY: 2M Delay: 9990 us
AdvMode: Connectable
AdvDataInfo: A1 02
47 07 46 18 A1 02 08 4D 21
G     F              M  !

Timestamp: 12.876234    Length: 45      RSSI: -65       Channel: 8      PHY: 2M
Ad Type: AUX_ADV_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 43
AdvMode: Connectable
AdvA: 5B:D8:25:52:89:4C (RPA) AdvDataInfo: A1 02
47 2B 49 09 4C 89 52 25 D8 5B A1 02 02 01 02 08 09 50 69 78 65 6C 20 38 02 0A F9 11 07 42 00 74 A9 FF 52 10 9B 33 49 35 9B 00 04 68 EF
G  +  I     L     R  %     [                       P  i  x  e  l     8                 B     t        R        3  I  5           h   
~~~

I'm not sure why your firmware isn't correctly receiving on the long range coded PHY. I've noticed that sometimes `cmd_chan_aa_phy` at the start of running the sniffer can fail, though just launching `sniff_receiver.py` again with the same options should fix it. If that doesn't fix it, you'd need to debug why the `cmd_chan_aa_phy` (to switch to the coded PHY) is failing.

froloffw7, 2024-04-07:
Thank you for the quick response.

I believe there is some problem with the UART programming of my module. I'll try to investigate more deeply.

This is definitely not a hardware problem, as I'm using BSL for board reprogramming.

In the meantime, I did some tests by forcibly assigning `phy = PHY_CODED_S8;` at the _RadioWrapper_recvFrames()_ function entry. But I didn't receive any output from the module.

froloffw7, 2024-04-07:
PS. I successfully rebuild the project with:

-  simplelink_cc13xx_cc26xx_sdk_7_40_00_77
- sysconfig_1.19.0
- arm-gnu-toolchain-13.2.Rel1

with only minor changes related to switch from UART to UART2 TI driver.

froloffw7, 2024-04-09:
Dear Sultan Qasim Khan,

Thank you for this fantastic project. It's been a valuable tool in diagnosing malfunctions.

While working with it, I made some minor changes that I believe could also be helpful.

Changes:

- add AUX_CONNECT_RSP
- packet hex dump
The output now looks like:
```
Timestamp: 12.500241    Length: 9       RSSI: -61       Channel: 37     PHY: Coded (S=8)
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 0 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 3 PHY: Coded Delay: 6480 us
AdvMode: Connectable
AdvDataInfo: 01 00
00000000  07 07 46 18 01 00 03 d8  40                       |..F.....@       |
00000009

Timestamp: 12.506750    Length: 257     RSSI: -73       Channel: 3      PHY: Coded (S=8)
Ad Type: AUX_ADV_IND
ChSel: 0 TxAdd: 0 RxAdd: 0 Ad Length: 255
AdvMode: Connectable
AdvA: 66:55:44:33:22:11 (Public) AdvDataInfo: 01 00
00000000  07 ff 49 09 11 22 33 44  55 66 01 00 02 01 06 1a  |..I.."3DUf......|
00000010  ff 4c 00 02 15 fd a5 06  93 a4 e2 4f b1 af cf c6  |.L.........O....|
00000020  eb 07 64 78 25 27 74 6b  ed c5 00 00 00 00 00 00  |..dx%'tk........|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000100  00                                                |.               |
00000101

Timestamp: 12.523941    Length: 36      RSSI: -61       Channel: 3      PHY: Coded (S=8)
Ad Type: AUX_CONNECT_REQ
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 34
InitA: 79:12:20:DA:A3:5C (RPA) AdvA: 66:55:44:33:22:11 (Public) AA: 0x66D796F2 CRCInit: 0xA54BD0
WinSize: 1 WinOffset: 12 Interval: 36 Latency: 0 Timeout: 500 Hop: 6 SCA: 5
Channel Map: FF FF FF FF 1F (all channels)
00000000  45 22 5c a3 da 20 12 79  11 22 33 44 55 66 f2 96  |E"\.. .y."3DUf..|
00000010  d7 66 d0 4b a5 01 0c 00  24 00 00 00 f4 01 ff ff  |.f.K....$.......|
00000020  ff ff 1f a6                                       |....            |
00000024

Timestamp: 12.527042    Length: 16      RSSI: -73       Channel: 3      PHY: Coded (S=8)
Ad Type: AUX_CONNECT_RSP
ChSel: 0 TxAdd: 0 RxAdd: 1 Ad Length: 14
AdvMode: Non-connectable, non-scannable
AdvA: 66:55:44:33:22:11 (Public) TargetA: 79:12:20:DA:A3:5C (RPA)
00000000  88 0e 0d 03 11 22 33 44  55 66 5c a3 da 20 12 79  |....."3DUf\.. .y|
00000010
```
If you'd be interested, I'd be happy to share these changes. Please let me know the preferred method for contribution, such as a pull request (PR) or a diff.

sultanqasim, 2024-04-09:
Thank you for your work. Please submit two separate PRs for the two features. Receipt and proper handling of AUX_CONNECT_REQ is something that was on my TODO list, so thanks again!

closed 2024-04-09 by sultanqasim