nccgroup/Sniffle #72: Full packet not written to pcap file

mkabjian, 2024-05-05:
Hello, thank you for such a useful library!

I am encountering an issue where I am not receiving full packet data when using Sniffle. I am attempting to capture Opendroneid messages transmitted via Bluetooth 5 and BLE using a standard Opendroneid Bluetooth transmitter. However, the packets captured in the resulting PCAP file and displayed in the console output are only 33 to 56 bytes, despite the expected packet size from my transmitter being 271 bytes.

Here is the command I am using:
./sniff_receiver.py -s /dev/ttyUSB0 -e -o bluetooth_test.pcap

sultanqasim, 2024-05-05:
I’m not sure how 271 byte packets would be transmitted. The maximum PDU size for BLE is 255 bytes plus a 2 byte header and 3 byte CRC (not included in Sniffle’s output but it is checked by the hardware). Could you share a PCAP of what you’re seeing?

mkabjian, 2024-05-05:
Thank you for your response.  Yes I was just looking into this.  I thought BT5 extended advertisements were larger based on the Opendroneid spec and sample data.  My guess is the transmitter implementation is chunking the data into smaller advertisements. 

I apologize for erroneously opening an issue. I will work on decoding the packets and confirm this is a transmitter issue and not Sniffle. Thanks again for your great work!

sultanqasim, 2024-05-05:
If it's using BT5 extended advertising, you could achieve longer advertising with AUX_CHAIN_IND PDUs, for example doing a 255 byte chunk followed by a 16 byte chunk. Is the sniffer properly following AuxPtrs in ADV_EXT_IND and AUX_ADV_IND PDUs you see? You're using the `-e` flag with the sniffer?

mkabjian, 2024-05-06:
This was not problem.  It was just a misunderstanding of the proper packet size from the opendroneid spec. We have the packets decoding fine now.

I was using -e and -e -l with the sniffer and the packets look good.

closed 2024-05-06 by mkabjian