nccgroup/Sniffle #76: Please add support for Sonoff Zigbee 3.0 USB Dongle Plus V2

ZeroChaos-, 2024-05-17:
When I plug in my device I see this in dmesg
```
New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.42
New USB device strings: Mfr=1, Product=2, SerialNumber=3
Product: SONOFF Zigbee 3.0 USB Dongle Plus V2
Manufacturer: ITEAD
ttyACM0: USB ACM device
```
Unfortunately trying to flash it simply fails
```
zero@naga /tmp % sudo cc2538-bsl.py -p /dev/ttyACM0 --bootloader-sonoff-usb -ewv /usr/share/Sniffle/sniffle_cc1352p1_cc2652p1_1M.hex
sonoff
Opening port /dev/ttyACM0, baud 500000
Reading data from /usr/share/Sniffle/sniffle_cc1352p1_cc2652p1_1M.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

I ordered this hardware from Ameridroid in April 2023 if that helps.  I've literally never used it until trying to flash it with your Sniffle firmware :-)

ZeroChaos-, 2024-05-17:
I am using this fork of cc2538-bsl.py if it helps to know that https://github.com/sultanqasim/cc2538-bsl

sultanqasim, 2024-05-17:
Hmm, the error when flashing means it's failing to enter the bootloader. Is your dongle CC2652P based or EFR32 based? I read that the "V2" dongle is the EFR32-based ZBDongle-E, whereas Sniffle needs the CC2652P based ZBDongle-P.

ZeroChaos-, 2024-05-18:
I opened this before I realized it was a fully different chipset.  The v2 is the -E which is EFR32.  If that chip isn't close enough to consider porting then please feel free to close this.

sultanqasim, 2024-05-18:
Unfortunately the radio hardware in that chip is completely different, so it’s not compatible.

closed 2024-05-18 by sultanqasim