nccgroup/Sniffle #77: (Question) Sniffle ubertooth-specan-ui port

alphafox02, 2024-05-18:
The thought occurred to me that perhaps a port of this would be useful if it was compatible with sniffle.

https://github.com/greatscottgadgets/ubertooth/blob/master/host/python/specan_ui/ubertooth-specan-ui

[image]

I can think of at least one useful program that relied on this and the use of the Ubertooth, but with the Ubertooth seemingly no longer being sold, perhaps sniffle could step in. 

Program I’m thinking of that also utilizes hardware for Bluetooth pieces. 
https://github.com/ghostop14/sparrow-wifi?tab=readme-ov-file#ubertooth-one

sultanqasim, 2024-05-18:
The CC26xx hardware does support being used as a spectrum analyzer  (tuning to different frequencies and measuring RSSI), and TI even made an example program to do this for the CC1350 if I remember correctly. It even had a pretty decent range of frequencies across multiple bands in the case of the CC1350: <https://www.ti.com/tool/TIDC-01004>

I’ve contemplated implementing this, either as a feature in Sniffle or as a standalone firmware. TI’s sample firmware and desktop software for this is BSD licensed, and the desktop software is written with Qt, so making an open source and Linux friendly version that is compatible with the CC1352, CC1354, CC2651, CC2652, and CC2674 should definitely be feasible. The CC26xx devices would be limited to roughly 2.2-2.6 GHz (not supporting the sub-1 GHz bands) but that would still be useful.

As for integtrating it with Sniffle vs. keeping it as a separate firmware, I’m not yet decided. Adapting the TI firmware without integrating it into Sniffle would probably be less work, and it would work with TI’s host side software (which I could adapt to work with Qt 6). If I did that, I’d still try to bring the firmware build process (makefile, linker scripts, build scripts, GCC usage) in line with Sniffle.

Spectrum analysis is somewhat secondary to Sniffle’s main purpose (Bluetooth sniffing and transmission with low level control). However, having it built into the same firmware could be convenient. Maybe I could consider integrating it into Sniffle after I get TI’s own spectrum analyzer firmware working on the newer chips.

On a separate note, check out my improvements to SigDigger’s panoramic spectrum feature and my fast retuning RTL-SDR driver. While the RTL-SDR doesn’t support the 2.4 GHz band, it’s quite slick for the bands it does support 😁. Or if you use my customized fast retuning USRP B2xx driver, you can get it to sweep really fast and look at bands up to 6 GHz.

alphafox02, 2024-05-20:
I had no idea you did a mod for SigDigger, I’ve got to go check this out. I have thd rtlsdr and b210 to test with. I think in newer releases SigDigger went to just soapy interfaces to the hardware.

I think the spectrum view w/ sniffle may be useful to some, just wasn’t sure how widespread it’d be used.

sultanqasim, 2024-05-20:
I’ve been contributing to SigDigger on and off for the past year. The modified panoramic spectrum might give an error when stopping it on Linux, but we’ll fix it.

sultanqasim, 2024-05-28:
Inspiration: https://github.com/sultanqasim/ti_sa13x0

alphafox02, 2024-05-28:
I was thinking about what you said 

“ The CC26xx devices would be limited to roughly 2.2-2.6 GHz (not supporting the sub-1 GHz bands) but that would still be useful.”

Say you had an RTLSDR/Airspy, it probably would be pretty handy to have the sniffle spec-a like ability to cover the area you can’t really get to with those devices at least to visualize I guess what the spectrum is looking like?