nccgroup/Sniffle #78: Questions, Requests - Quiet parameter in Wireshark, Delta time column populated, Custom GATT dissection

phgarman, 2024-05-23:
Really like this project.

**### Quiet Parameter**
I've noticed that when I run the sniffer from the command line using the -q option, it seems to filter the Empty PDU frames as expected in the command window, but those still get inserted into the PCAP output when viewing in Wireshark.  Is that intended?  It would be nice if the Empty PDU's were excluded in the PCAP.  Also would be even better if when I run directly from Wireshark, the config options had a checkbox option to exclude those.

**### Delta Time Column**
Would also be nice if the delta time between frames was populated into the column.

**### Custom GATT Dissection**
This is probably more of a Wireshark dissector thing, but the BLE devices I'm working with use custom characteristics.  For example, Rcvd Read Blob Response, Handle: 0x0014 (Unknown).  Is it possible to define Handle interpretations from the extcap script or would this need to be done via a dissector of sorts?

sultanqasim, 2024-05-23:
Regarding the `-q` parameter, that is indeed the intended behaviour. In Wireshark, you can set up a display filter to hide empty PDUs, such as: `btle.length > 0`

Regarding displaying delta times, you can do that in Wireshark by adding a column of type "Relative time".

For custom GATT dissection, that would need to be done via some sort of dissector. The extcap script only feeds the raw PDUs into Wireshark.

sultanqasim, 2024-06-03:
Has your question been answered?

phgarman, 2024-06-03:
Yes, thank you

closed 2024-06-03 by phgarman