nccgroup/Sniffle #84: Trouble with receiving 2m and coded phy

IakovlevAA, 2024-07-10:
Hi, I'm using TI cc2652-r7 dev.kit with prebuilt firmware from release(sniffle_cc2652r7.hex). I can successfully catch all 1m ble packets, but using
 ```sudo ./sniff_receiver.py -le```
Gives no result at all. I'm not having erorrs, so can't attach any log, Any guesses where to search for problem?

IakovlevAA, 2024-07-10:
I've noticed that in Readme you are using firmware_3.0.0.28_fastuart.bin. But when flashing sniffle_cc2652r7.hex with uniFlashGUI.sh it automatically updated to 3.0.0.29. Could this cause an issue?

sultanqasim, 2024-07-10:
The XDS110 UART firmware modification is only needed when you care about having low receive latency. The sniffer will work fine with the regular XDS110 firmware, just with variable latency when receiving packets, up to a few hundred ms.

When you run the command `./sniff_receiver.py -le`, it will listen for extended advertising on the long range PHY, and nothing else. Most devices do not use the long range PHY, and so usually you will see no data unless you have something transmitting advertisements on the long range PHY. You can test that out by setting up an advertiser using the long range PHY with an Android app like nRF connect.

IakovlevAA, 2024-07-10:
Thanks! I just used wrong phy mode while setting nrf connect advertiser...

closed 2024-07-10 by IakovlevAA