nccgroup/Sniffle #85: Relay

johnb21098, 2024-07-15:
Hi, graet product. I ahve bought some TI dev boards to play around with. 

Have you documented the relay attack methodology anywhere? I'm looking to give it a go and didn't know whether there were some documented instructions somewhere? Or if you could give us a point in the right direction. 

Thanks!

johnb21098, 2024-07-15:
Many thanks for your quick reply. 

By the host side glue code, is this the part of sending the BLE data between two of the sniffle devices and ensuring its transmitted? That's what I was wondering how to implement properly. 

Many thanks

sultanqasim, 2024-07-15:
Yes, that’s what I meant by host side glue.

johnb21098, 2024-07-16:
are you able to say if we can just transmit the pcap files we get or do we need to do some processing in the middle?

sultanqasim, 2024-07-21:
It’s mostly just forwarding packets from one side to another, though you do need to keep track of the connection event counter on both sides. If they’re too far out of sync, there will be issues with connection parameter changes. For unencrypted connections, you can change the instant value in connection parameter change requests to avoid issues. For encrypted connections, you need to keep the connection event counters roughly in sync.