nccgroup/Sniffle #86: Ext cap not detected

JahazielLem, 2024-07-15:
Hello! 
I had the Version 4.2.6 of Wireshark running on M1 macbook
[image]
I follow the instruction in the README, using a symlink
```
mkdir -p ~/.local/lib/wireshark/extcap
ln -s $(pwd)/python_cli/sniffle_extcap.py ~/.local/lib/wireshark/extcap
```
[image]

But still dind't working
[image]
I change the head to use /opt/homebrew/bin/python3 and install pyserial as global library, in linux I had the same problem, in windows ok

closed 2024-07-15 by JahazielLem

lukeswitz, 2024-08-11:
Same issue, did you solve this on unix systems?

lukeswitz, 2024-08-11:
I found my issue, instead of a homebrew path it just needed a venv path specified with pyserial in there also. Cheers