nccgroup/Sniffle #87: initiator.py cannot connect target

pic7oris, 2024-07-19:
Hi, when I use initiator.py to connect to a target, some of the targets connect successfully and keep getting LL interactions, but most of the devices don't connect successfully, and I'm using another sniffle to sniffler the broadcast messages, and I don't see the initiator.py emitting anything!

```
ϟ  Sniffle_20240606/python_cli master ✗  python3.9 initiator.py -m 08:3a:f2:b9:83:c2
Timestamp: 0.022365  Length: 39  RSSI: -40  Channel: 37  PHY: 1M  CRC: 0xD4E355
Ad Type: ADV_IND
ChSel: 0 TxAdd: 0 RxAdd: 0 Ad Length: 37
AdvA: 08:3A:F2:B9:83:C2 (Public)
0x0000:  00 25 c2 83 b9 f2 3a 08  02 01 06 0f 09 45 53 50  .%....:......ESP
0x0010:  33 32 5f 42 4c 45 5f 47  41 54 54 05 03 ee 00 ff  32_BLE_GATT.....
0x0020:  00 05 12 06 00 10 00                              .......

TRANSITION: INITIATING from STATIC

```

sultanqasim, 2024-07-27:
This target has a public MAC address. You need to use the `-P` argument for connecting to targets with a public MAC.

pic7oris, 2024-07-29:
> This target has a public MAC address. You need to use the `-P` argument for connecting to targets with a public MAC.

wow！Success！Think you, This is a great tool.

closed 2024-07-29 by pic7oris