nccgroup/Sniffle #88: AttributeError: module 'numpy' has no attribute 'typing' (rfnm)

alphafox02, 2024-08-02:
Just adding mention here that on 22.04 with the apt install version of numpy, the following occurs when trying to use the -s rfnm argument.

AttributeError: module 'numpy' has no attribute 'typing'

Uninstalling the apt numpy would remove gpsd-clients, kismet, and a lot of other applications built around the 22.04 apt version of python3-numpy. 

Would it be possibly to work the -s slightly different? It seems like in numpy 1.2+ that it should work, but I'm not too familiar.

wuzhongjie1997, 2024-10-23:
I'd suggest that you run the Sniffle python interface in a dedicated virtual environment, e.g. using python venv module, conda or the like. Python packages are (imho) notoriously hard to stay compatible for long periods of time, and setting up multiple venv's (especially using conda) really simplifies things a lot for me.

alphafox02, 2024-10-23:
Totally agree. I forget what I ended up doing but it was something along the lines of removing one line that wasn’t really needed and then it ran fine even without the venv. I think it was just for type hints/notes.