nccgroup/Sniffle #91: Extended advertising capture failing after a while with multiple advertisers

sultanqasim, 2024-08-22:
After a while of sniffing multiple extended advertisers, the sniffer stops following auxiliary pointers for any of the devices. This may take several minutes to happen. I reproduced this issue (originally reported to me by @alphafox02) by running multiple smartphones as long range extended advertisers with long-ish (100+ byte) auxiliary advertisements. After around 12 minutes, it stopped following extended advertising. 

I still need to find the root cause. I'm guessing it's a bug in the auxiliary advertising scheduler, and it may or may not involve wraparound of the 32-bit radio timer.

sultanqasim, 2024-08-22:
I triggered the bug again, this time in less than a minute, and the 32-bit radio timer was nowhere near wraparound, so it's probably unrelated to wraparound of that.

sultanqasim, 2024-08-22:
Last few ads where this happened:

```
raw ts 0x755da8b
Timestamp: 57.527382  Length:  9  RSSI: -58  Channel: 37  PHY: Coded (S=8)  CRC: 0x94E31B
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 15 PHY: Coded Delay: 4980 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0x8c2 SID: 0x0
0x0000:  47 07 06 18 c2 08 0f a6  40                       G.......@

raw ts 0x755ee14
Timestamp: 57.532383  Length: 124  RSSI: -58  Channel: 15  PHY: Coded (S=8)  CRC: 0x25E7C7
Ad Type: AUX_ADV_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 122
AdvMode: Non-connectable, non-scannable
AdvA: 58:F5:18:5C:D2:97 (RPA) AdvDataInfo DID: 0x8c2 SID: 0x0
Complete Local Name: Pixel 6
Tx Power Level: -7 dBm
Complete List of 16-bit Service Class UUIDs
    0x1901
Service Data - 16-bit UUID
    Service: 0x1234
    Data Length: 92
    Data: b'\xf0\r\xfa\xce\xca\xfe\xba\xbe\xda\xdb\xee\xff\x00\xdf\xac\xec\xaf\xeb\xab\xed\xad\xbe\xef\xf0\r\xfa\xce\xca\xfe\xba\xbe\xda\xdb\xee\xff\x00\xdf\xac\xec\xaf\xeb\xab\xed\xad\xbe\xef\xf0\r\xfa\xce\xca\xfe\xba\xbe\xda\xdb\xee\xff\x00\xdf\xac\xec\xaf\xeb\xab\xed\xad\xbe\xef\xf0\r\xfa\xce\xca\xfe\xba\xbe\xda\xdb\xee\xff\x00\xdf\xac\xec\xaf\xeb\xab\xed\xad\xbe\xef'
0x0000:  47 7a 09 09 97 d2 5c 18  f5 58 c2 08 08 09 50 69  Gz....\..X....Pi
0x0010:  78 65 6c 20 36 02 0a f9  03 03 01 19 5f 16 34 12  xel 6......._.4.
0x0020:  f0 0d fa ce ca fe ba be  da db ee ff 00 df ac ec  ................
0x0030:  af eb ab ed ad be ef f0  0d fa ce ca fe ba be da  ................
0x0040:  db ee ff 00 df ac ec af  eb ab ed ad be ef f0 0d  ................
0x0050:  fa ce ca fe ba be da db  ee ff 00 df ac ec af eb  ................
0x0060:  ab ed ad be ef f0 0d fa  ce ca fe ba be da db ee  ................
0x0070:  ff 00 df ac ec af eb ab  ed ad be ef              ............

raw ts 0x7574093
Timestamp: 57.619038  Length:  9  RSSI: -36  Channel: 37  PHY: Coded (S=2)  CRC: 0xA1E025
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 24 PHY: Coded Delay: 9990 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0xc31 SID: 0x1
0x0000:  47 07 06 18 31 1c 18 4d  41                       G...1..MA

raw ts 0x7574da5
Timestamp: 57.622384  Length:  9  RSSI: -58  Channel: 37  PHY: Coded (S=2)  CRC: 0x2E29C7
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 15 PHY: Coded Delay: 11250 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0xbc2 SID: 0x1
0x0000:  47 07 06 18 c2 1b 0f 77  41                       G......wA

raw ts 0x75767a5
Timestamp: 57.629040  Length: 178  RSSI: -39  Channel: 24  PHY: Coded (S=2)  CRC: 0x8DF1D0
Ad Type: AUX_ADV_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 176
AdvMode: Non-connectable, non-scannable
AdvA: 4C:96:AA:56:6E:21 (RPA) AdvDataInfo DID: 0xc31 SID: 0x1
Complete Local Name: Pixel 8
Manufacturer Specific Data
    Company: 0x1234
    Data Length: 64
    Data: b'\xaa\xaa\xaa\xaa\xbb\xbb\xbb\xbb\xcc\xcc\xcc\xcc\xdd\xdd\xdd\xdd\xaa\xaa\xaa\xaa\xbb\xbb\xbb\xbb\xcc\xcc\xcc\xcc\xdd\xdd\xdd\xdd\xaa\xaa\xaa\xaa\xbb\xbb\xbb\xbb\xcc\xcc\xcc\xcc\xdd\xdd\xdd\xdd\xaa\xaa\xaa\xaa\xbb\xbb\xbb\xbb\xcc\xcc\xcc\xcc\xdd\xdd\xdd\xdd'
Tx Power Level: -7 dBm
Complete List of 16-bit Service Class UUIDs
    0x1802 (Immediate Alert)
Service Data - 16-bit UUID
    Service: 0x183B (Binary Sensor)
    Data Length: 78
[... 252 lines cut ...]
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 34 PHY: Coded Delay: 9990 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0xad1 SID: 0x0
0x0000:  47 07 06 18 d1 0a 22 4d  41                       G....."MA

raw ts 0x76111b0
Timestamp: 58.262395  Length:  9  RSSI: -58  Channel: 37  PHY: Coded (S=2)  CRC: 0x601278
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 20 PHY: Coded Delay: 91230 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0xbc2 SID: 0x1
0x0000:  47 07 06 18 c2 1b 14 e1  4b                       G.......K

raw ts 0x7612538
Timestamp: 58.267395  Length:  9  RSSI: -58  Channel: 37  PHY: Coded (S=2)  CRC: 0xAA737E
Ad Type: ADV_EXT_IND
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 7
AuxPtr Chan: 17 PHY: Coded Delay: 105000 us
AdvMode: Non-connectable, non-scannable
AdvDataInfo DID: 0x8c2 SID: 0x0
0x0000:  47 07 06 18 c2 08 11 ac  4d                       G.......M
````

sultanqasim, 2024-08-23:
I figured out the root cause. The bug is in DelayStopTrrigger and its usage of timers. The timer always counts up, and it only gets loaded to zero on initialization or on match. Calling `Timer_stop()` or `Timer_start()` does not reload the timer's counter. Consequently, if the timer was set to count up to some value, then we need to move the trigger point to some time sooner, we might end up setting the timer match value to something below the current time value. In that case, the timer will only trigger after the counter wraps around.

According to <https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/710802/launchxl-cc1352r1-gptimercc26xx---resetting-running-load-value-to-zero> there is no API to manually reload timers, and TI recommends using the `ClockP` API instead.

alphafox02, 2024-08-23:
With the changes and a new firmware built I’ve so far been unable to reproduce the bug. I turn on two bt5 long range extended transmitters and both devices are tracked and decoded. I’ll continue to run for longer periods of time and try to introduce more of the same type of transmitters. Thank you for addressing this, seems to work great now!

sultanqasim, 2024-08-23:
Fixed in 30cfe09