nccgroup/Sniffle #104: Removing simplelink SDK directory once sniffle built

D3vil0p3r, 2024-11-16:
Once I built Sniffle by running `make` in `fw` folder, if everything is ok, can I then remove `simplelink_cc13xx_cc26xx_sdk_7_41_00_17`, `sysconfig` and `xdctools_3_62_01_15_core` directories?

sultanqasim, 2024-11-17:
The SDK is only needed for building the firmware. It's not a runtime dependency. You don't need it if you're using prebuilt firmware or you already built the firmware yourself.

D3vil0p3r, 2024-11-17:
> Uniflash is only a build dependency or a runtime dependency for Sniffle?
>
What about Uniflash dependency? Is it runtime or build dep?

sultanqasim, 2024-11-17:
It’s only used for flashing the firmware. You don’t need Uniflash at runtime.

D3vil0p3r, 2024-11-17:
So I guess we don't need at building time too...

sultanqasim, 2024-11-17:
Correct, you don’t need Uniflash to build the firmware.

closed 2024-11-17 by D3vil0p3r