nccgroup/Sniffle #105: Sniff all packets on all channels?

johnboy00, 2024-11-29:
I'm interested in being able to sniff on all 40 channels (adv + data).  Missing some packets while channel hopping is of no consequence, and I don't care about the type of packet or its payload.  I only care about the source and destination MACs.  Your python CLI doesn't appear to support this, but can it be modified to support it?  If so, can you give me any hints as to where and how?  Is the current f/w able to do this?  I'm using a SONOFF dongle (CC2652P).

Thanks!
J

sultanqasim, 2024-11-29:
For sniffing advertisements, this is fairly straight forward to achieve with the existing firmware and Python APIs. See the [cmd_chan_aa_phy](https://github.com/nccgroup/Sniffle/blob/9c57a7169932622db119201956bb692fa5c24f6b/python_cli/sniffle/sniffle_hw.py#L135) command in particular. You can have a Python script periodically issue this command cycling between channels (and PHY modes if you'd like) while using the access address of 0x8E89BED6 (`BLE_ADV_AA`) and initial CRC of 0x555555 (`BLE_ADV_CRCI`).

For sniffing data packets, this is a bit trickier. Specifically, all data packets start with a random connection-specific access address that is not 0x8E89BED6. The radio hardware uses the access address as a sync word, and without a known sync word it doesn't know where packets start. If you know the access address of a connection, you can use `cmd_chan_aa_phy` with that access address, and just disable CRC validation if the initial CRC is unknown.

If you don't know the connection's access address, and/or want to capture packets from all ongoing connections regardless of their access address, then it becomes considerably more difficult. It is possible to synchronize on the eight bit preamble of alternating ones and zeros, but this is unreliable and results in many false synchronizations and off-by-two bit errors. The Ubertooth supports this approach to recover access addresses, though as I said its error-prone with many false syncs and missed syncs. I had an intern implement this on the CC26x2 for me some years ago, but it wasn't working very well and I had not put in time to try to improve its performance.

One other possibility is Software Defined Radio (SDR) based sniffing. I've been working on SDR support for Sniffle with [RFNM](https://rfnm.io/) though that is a work in progress. The RFNM hardware is capable of sniffing everything happening on all 40 channels simultaneously, and SDR allows more control over signal processing, so it would be easier to achieve usable levels of performance/reliability with synchronization on the preamble.

johnboy00, 2024-11-29:
Thank you so much for the quick and thorough response!