nccgroup/Sniffle #108: Sending packets requiring Instants

XenoKovah, 2025-01-22:
If I'd like to send an LL_PHY_UPDATE_IND to a peripheral to change to 2M PHY, I need to set an Instant when that will take place. What is the best way to do this? I'm guessing that querying the connEventCount from the python code and setting some value at some point in the future would either be too slow, or would require a bunch of binary search trial and error to get to the right minimum reliable Instant. Is there (/ should there be) a way to ask the firmware to fill in a "current connEventCount + X" value where the python side can set the X to 6 or 10 or whatever it wants?

sultanqasim, 2025-02-05:
I don't have a feature for that right now, though it would be nice to have. You can estimate the current connEventCount based on time since connection start or the connEventCount value in the last packet received from the sniffer. Changes to connEventCount aren't too fast (minimum connection interval is 7.5 ms), so using the last received connEventCount + 8 or 9 should work.

XenoKovah, 2025-04-24:
I've got some time to try this now, but I'm trying to understand the second half of your comment "You can estimate the current connEventCount based on time since connection start or the connEventCount value in the last packet received from the sniffer.", specifically the "or the connEventCount value in the last packet received from the sniffer." 

Are you saying there's already a way to get the connEventCount for the last received packet on the Python side? 

Or were you just saying that's one easyish thing that could be added to the code to support packets that require instants?

sultan-ncc, 2025-04-24:
I meant that connEventCount is in the `event` field of the `PacketMessage` class. See https://github.com/nccgroup/Sniffle/blob/master/python_cli/sniffle/packet_decoder.py#L77

XenoKovah, 2025-04-27:
Thanks for pointing out that field. I had looked at the `PacketMessage` class definition but the `event` field didn't jump out at me (though in retrospect I understand the naming now.) FWIW I had already implemented a time-based approach by the time you mentioned that, and it worked, but then I replaced it with usage of `event` which also works. Overall this seems to be working, although it turns out that my test Zephyr device behaves in a way that my code's hackish state machine couldn't originally handle (with the Peripheral initiating packets it didn't previously, after the PHY change), so I'm having to rework and rewrite it significantly to get it back into a better state. But I'm pretty sure Sniffle's handling things fine so I'm going to close this. Thanks again.

closed 2025-04-27 by XenoKovah