nccgroup/Sniffle #109: Hardware Compatibility Question

iwm911, 2025-02-10:
Hi,

Thank you for your great work on this project.

It has become harder to source the Sonoff -P version dongle, do you think this board might work?
will the firmware require modifications for it to work?

https://www.aliexpress.com/item/1005007564063140.html

Thanks.

sultanqasim, 2025-02-10:
It should work if the pinout matches the corresponding Launchpad and Sonoff boards. The main things that matter are serial pins, the LED, and the bootloader pin. The schematic linked from that page is a bit strange with the things like the DIO5 pin supposedly connected to DIO1 signal, and the CC2652P being labelled as a CC1352P. As such, I can't make any guarantees about that board working when the schematic doesn't make sense, but in general if that board works with firmware for corresponding Launchpad boards, then it will probably work with Sniffle. The pinout and compatibility table here also applies to Sniffle: https://github.com/Koenkk/Z-Stack-firmware/issues/210

XenoKovah, 2025-02-16:
@iwm911 since I'm going to be asking folks to buy Sonoffs for a future BLE class, is there some reason why sourcing from the locations at [Sonoff's website](https://sonoff.tech/product/gateway-and-sensors/sonoff-zigbee-3-0-usb-dongle-plus-p/) doesn't work? (This would help me know if there's geographic regions specifically where it's currently problematic.) Thanks

alphafox02, 2025-03-11:
I buy them in bulk, easy to get. Haven’t had an issue yet.

sultanqasim, 2025-03-11:
In Canada, I can get them from Sonoff directly easily. However, they're no longer available on Amazon in Canada.